opentrack/opentrack #343: Motionjoy DS3 Tool + Opentrack's Vjoy Issue???

benemii, 2016-04-24:
So i'm having this "issue" where i cant get just  Opentracks Vjoy to output in to a game (Elite Dangerous)
So long story short i made a custom controller layout on MotionJoy's DS3 Tool for my DualShock3 Controller because i wanted my R2 and L2 Buttons to be my Z axis as they once were on DS3 because i wanted those two to control my Yaw.

It works by default when i only use my DS3 Tool in Elite Dangerous without a single problem.

Except.. It was a tad too sensitive.  So i had the bright idea to pump that info through OpenTrack using the Joystick input(Set to MotionJoy) and use the Axis curves to smooth the output out and it Worked! ... Except it ONLY worked for the "Odd" axis' that weren't assigned to the Normal ones( X axis Z axis Y axis). 

So what im saying is in Elite Dangerous my normal Pitch Yaw and Roll axis are being controller by the DS3 Tool instead of Vjoy through OpenTrack. It only works when i change a normal axis (Let say my default pitch axis) to an "odd" one like Rx+ which should be my Right Stick. (I normally use my Left stick for pitch). When i change it in Open track to that "odd"(meaning not the usual axis) axis' itl work and register in Elite Dangerous as the Vjoy axis. But the problem is just that. It ONLY works for those "odd" Axis. It wont work for the normal axis. Elite will just use the unsmooth DS3 Tool Axis and not the Smoothly Adjusted using the Axis Curves in OpenTrack,Vjoy Axis.

Any tips on how to get this game to ONLY take input from OpenTrack's Vjoy output? Oh yeah
Sidenote; 
I don't see a Vjoy Device Listed in my Game Controllers Menu thats linked with OpenTrack's Vjoy in anyway. I only see One other Vjoy Device that i downloaded today. I'm also on Windows 7 btw.

Hope this wasnt too confusing. If it was,just in case here's the very very clear rundown;

Dualshock3 Controller is hooked up to my PC using bluetooth/usb.
That's going in to MotionJoy's DS3 Tool on a custom Profile.
Opentrack is open and set to Joystick Input (Set to input from the MotioJoy program)
Opentrack is set to output using its default Vjoy output

Elite Dangerous is open yet only takes input from Motiojoy's DS3 Tool unless a particular axis is set to an odd/normally not used axis such as Rx+  in Opentrack.

Help me jeezis?

sthalik, 2016-04-29:
@benemii use freetrack 2.0 for ED.

benemii, 2016-05-02:
@sthalik Uhhh You sure that will transfer my Movement controls over? I thought freetrack only contributes to the view movement ie:headmovement/headlook and not directly to ship movement?

sthalik, 2016-05-04:
Sorry, I've only skimmed your message.

If the vjoy controller isn't visible at all, check if it's visible in the system. opentrack won't notify you if the vjoy driver isn't installed at all unfortunately. There are two projects named "vjoy" and we're using the closed-source one. This is all legacy code and while works, there's been no new work done on virtual joystick output.

benemii, 2016-05-04:
@sthalik Does your version of Vjoy automatically install itself when i download/use OpenTrack? If not then is there any way you can tell me how a normal person is supposed to get your version of vjoy to work? When it worked for that one Axis,it worked wonders,No jittering(ingame),no drifting,everything was as smooth the way it was supposed to be. So if its not too much to ask,it would be great to get this to work.

sthalik, 2016-05-04:
@benemii no, it doesn't install itself. Only the files opentrack itself needs are included. You need to install the driver manually.

This is the driver - http://www.headsoft.com.au/index.php?category=vjoy

They have a big stupid animated logo, scroll down.

trev0rbr, 2016-05-07:
Don´t use MotionJoy .

Latest drivers alternative project http://emulation-general.wikia.com/wiki/SCP_Driver_Package

closed 2016-07-19 by sthalik