opentrack/opentrack #344: Been messing with ps3 eye w/ filter cant find center of my head.

LaChapel, 2016-04-24:
My middle led on my track clip is registering as the center of my head and as the led. The Middle led has a red x and a green x. I have messed with everything i dont know what to do. Someone help. Oh and i can confirm all three led's are active and all at the same brightness.
[image]

Len62, 2016-04-24:
With program running click on the model tab and at the bottom do the calibration. There is link on that tab to take you to the wiki re calibration.

LaChapel, 2016-04-24:
I have tried that twice now and it cant manage to find the center of my head. Thanks for the help, But sadly didn't help.

LaChapel, 2016-04-24:
its weird its registering as red and green. so it thinks thats the center of my head as well has the middle LED.

Len62, 2016-04-24:
Yes, the red is the center of your head. Once you complete the calibration are you clicking the calibrate button again? That's what sets it.

LaChapel, 2016-04-24:
YESS you got it Thanks!!
Took me while but it worked. Now im having issues of not being able to get 120fps with the eye.

sthalik, 2016-04-25:
120 Hz and more is only for 320x240. But don't use it since lower resolution is more noisy.

sh

Len62, 2016-04-25:
I run it at 640x480 at 60Hz and it works fine.

MathijsG, 2016-04-26:
Can't we change the calibration to some auto set duration? The clicking again is somewhat counter-intuitive.

sthalik, 2016-04-27:
We *can*, but should we?

From the nature of the statistical process the user should have control over when to stop supplying samples.

MathijsG, 2016-04-27:
But maybe we can guide them somewhat? If you have too few samples you just get a weird negative value currently.

sthalik, 2016-04-28:
It's not as much a matter of *too few* samples but *too similar* samples. The algorithm works fine if there's no ambiguity in input.

Also see that calibration settings link to the wiki instructions. There's not much else to do unless you can improve upon these instructions.

sthalik, 2016-04-28:
Also see that the calibration button is visibly pressed while calibration is running. It's implicit that the button is two-state like that.

MathijsG, 2016-05-19:
Today I gave one of my friends a trackclip device which I made and a PS3 webcam, and installed the thing with opentrack. For mor than half an hour we tried to calibrate, but it's just too hard. We constantly tried to do the yaw and pitch movement, required for calibration, but we all got those weird negative values. Last time we tried to do the movements very quick and luckily this time it worked. But it is kind of annyoing, not knowing wether your calibration is good enough already. Can't we somehow output the numbers already while calibrating? And when they are still those extreme negative values somehow explain this to the user while calibrating?

I don't really know the whole technical process of it, but I do see a lot of struggle with calibrating, from my own experience ánd others.

Len62, 2016-05-20:
@MathijsG 
I was experiencing the same negative value numbers occasionaly in the RC21p11 version, could never figure out why. In the latest version I haven't noticed it.

sthalik, 2016-05-20:
@MathijsG calibration values are fed on fixed interval. It's a fraction of a second. Maybe increase the PT calibration value somewhat?

MathijsG, 2016-05-20:
We already calibrated it in the end. But the procedure was quite frustrating, partly because of the lack of feedback the calibration system gives. I thought the longer, the more accurate. Which is wrong I know now. And what if during the movement on some weird occasion where you yaw your head a little too far, the system detects 4 leds, due to reflection, or maybe two leds. Do you need to start over then? Or can you just continue movement? And is the current orientation of the octopus relevant, or is it ignored completely? (I see Len62 centering the octopus before calibrating the thing, for example).

sthalik, 2016-05-20:
@MathijsG calibration is inaccurate if you supply the same pose too much. Then it's biased toward that pose as opposed to others.

If the octopus flips due to false detections at all then it's bound to affect calibration a lot.

Centering the octopus doesn't matter because it's done after PT and only PT affects calibration.

MathijsG, 2016-05-20:
Maybe we should state some more info on that calibration tab in Opentrack? Most of the calibration with tools and software seems to get more accurate while trying it longer. That's why I got confused.

sthalik, 2016-05-20:
We could give calibration unique poses only. That is, poses similar within 5-10 degres total could be discarded.

You can edit the wiki if you want. The calibration box links to the wiki.

sthalik, 2016-05-22:
Calibration now gets new values 4 times a second instead of 10. It should help. I'm closing the issue but you can reopen if it's still a problem to calibrate.

closed 2016-05-22 by sthalik