opentrack/opentrack #347: Razer Hydra orientation problem

LukyLK, 2016-04-27:
Hello, I've tried to use opentrack with my razer hydra as full motion headtrack for my homemade virtual reality. Positioning is fine but I have problem with orientatio. It seems that the axis input somehow affect together. Mostly pitch and roll. Any ideas how to fix? Thanks

sthalik, 2016-04-28:
In options there's a centering mode for cameras, switch to the other centering mode.

LukyLK, 2016-04-28:
I have got it, thats not problem. There is some problem on input. Yaw seems ok and in yaw=0 pitch and roll looks fine. But in yaw=90 pitch and roll changed together. It seem that there is some problem in translation around axis. All I write is abut raw tracker data, so it is some input problem I think.

sthalik, 2016-04-28:
Set your mappings to 1:1. Disable "center on startup". Don't center at all. See how the pose at yaw=90 corresponds to the device pose on the octopus image. Is it reasonable?

We may try something else - use our own matrix -> angle conversion function. There's a semantic difference from Euler and Tait-Bryan angle definitions.

But try the first listed thing, first.

sthalik, 2016-04-28:
If you're ever inclined about Tait-Bryan vs Euler, the definitions differ

    Proper Euler angles (z-x-z, x-y-x, y-z-y, z-y-z, x-z-x, y-x-y)
    Tait–Bryan angles (x-y-z, y-z-x, z-x-y, x-z-y, z-y-x, y-x-z).

so it's Tait-Bryan that is what we think of yaw, pitch, and roll.

LukyLK, 2016-04-28:
Yes I have 1:1 maping with no centering to see it but it dont care becase it is something wrong in raw tracker data. Is there any chance to switch the angel convergention function?

sthalik, 2016-04-28:
How does the octopus look like compared to the real-life device orientation?

I'm gonna give you a test build soon and now I have to leave.

sthalik, 2016-04-29:
@LukyLK try this https://www.dropbox.com/s/72rppmoay5mo1j8/opentrack-2.3-rc21p11-17-g3a7011f-dirty-20160429083657.zip

LukyLK, 2016-04-30:
Thats not it, unfortunately. Here is the video of the old version...
https://youtu.be/9xiI1c38KRI

LukyLK, 2016-04-30:
Thats not it, unfortunately. Here is the video of old version:
https://youtu.be/pVUMmEtg79s

and the new version:
https://youtu.be/ThCfqEnfCMY

sthalik, 2016-04-30:
I think that some axis are swapped in the new version. Can you remap the axis in options and check again? If it works I need a list of what to remap.

LukyLK, 2016-04-30:
Yaw ,pitch, roll is pitch, roll, yaw. But it doesnt work anyway.
https://youtu.be/aqsOUyw0huo

LukyLK, 2016-05-05:
Who is the creator of razer hydra plug in? Is it possible to get contact?

sthalik, 2016-05-05:
It's @mm0zct

mm0zct, 2016-05-05:
Hi LukyLK, I'm sorry I've not got an OpenTrack build environment set up at the moment, my home-office is still full of stuff from a kitchen rebuild so I've not had much PC time at home.

I think I always had this problem with the Hydra and OpenTrack, I never bothered because I was only using the positional aspect and combining that with rotation information from the Oculus Rift.

If I get a chance I'll get a new build env setup and take a look into it, but I can't make any promises, I've already got the Oculus 1.3 SDK update on my backlog.

LukyLK, 2016-05-05:
Time is always problem I understand it. But if will be some progres in the future tell me, I will be glad :-)

sthalik, 2016-05-06:
@mm0zct Oculus SDK 1.3 won't work on mingw without porting as usual. We should coordinate the porting and the tracker module effort.

Razer probably has some samples but I already have an opentrack backlog full of stuff including eye tracker samples to receive. My hands are full.

LukyLK, 2016-06-10:
I run it witf FreePie with this script

if starting:
    centerYaw = 0
    centerPitch = 0
    centerRoll = 0
    centerx = 0
    centery = 0
    centerz = 0
    centerYawa = 0
    
yaw = -(hydra[1].yaw )
pitch = (((math.cos(hydra[1].yaw-centerYawa))*(-hydra[1].pitch))+((math.sin(hydra[1].yaw-centerYawa))*(hydra[1].roll)))
roll = -(((math.sin(hydra[1].yaw-centerYawa))*(hydra[1].pitch))+((math.cos(hydra[1].yaw-centerYawa))*(hydra[1].roll)))
x = -hydra[1].x*0.7
y = hydra[1].y*0.8
z = hydra[1].z*0.8

freeTrack.yaw  = yaw - centerYaw
freeTrack.pitch = pitch - centerPitch
freeTrack.roll = roll - centerRoll

freeTrack.x  = x - centerx
freeTrack.y = y - centery
freeTrack.z = z - centerz

diagnostics.watch(hydra[1].pitch)
diagnostics.watch(hydra[1].yaw)
diagnostics.watch(hydra[1].roll)

if keyboard.getKeyDown(Key.LeftControl) and keyboard.getPressed(Key.C):
    centerYaw = yaw
    centerPitch = pitch
    centerRoll = roll
    centerx = x
    centery = y
    centerz = z
    centerYawa = hydra[1].yaw

It runs but FreePie dont run with some games. Opentrack do.
Is it possible to do the opentrack script with goniometric function like that?

sthalik, 2016-07-19:
@LukyLK what if you change `centering mode` in the main options window?

sthalik, 2016-09-26:
It's likely just gimbal lock. Let's leave it open for now.

closed 2018-01-18 by sthalik