opentrack/opentrack #349: Latest release

iDevPro, 2016-04-29:
```
Process:               opentrack.bin [1637]
Path:                  /Applications/opentrack.app/Contents/MacOS/opentrack.bin
Identifier:            com.github.opentrack
Version:               opentrack-2.3-rc21p11-14-gfb7e86d
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           opentrack.bin [1637]
User ID:               501

Date/Time:             2016-04-29 21:12:20.367 +0300
OS Version:            Mac OS X 10.11.4 (15E65)
Report Version:        11
Anonymous UUID:        F85BD5B7-4491-BB32-B224-3CC3F04AD619

Sleep/Wake UUID:       049C1B76-F9C0-498E-A96A-F5EC9CECCCC0

Time Awake Since Boot: 5800 seconds
Time Since Wake:       410 seconds

System Integrity Protection: enabled

Crashed Thread:        14  Tracker_PT

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
[1637] stack overflow

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff8eb7bf72 mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff8eb7b3b3 mach_msg + 55
2   com.apple.CoreFoundation      	0x00007fff984c21c4 __CFRunLoopServiceMachPort + 212
3   com.apple.CoreFoundation      	0x00007fff984c168c __CFRunLoopRun + 1356
4   com.apple.CoreFoundation      	0x00007fff984c0ed8 CFRunLoopRunSpecific + 296
5   com.apple.HIToolbox           	0x00007fff96fb0935 RunCurrentEventLoopInMode + 235
6   com.apple.HIToolbox           	0x00007fff96fb076f ReceiveNextEventCommon + 432
7   com.apple.HIToolbox           	0x00007fff96fb05af _BlockUntilNextEventMatchingListInModeWithFilter + 71
8   com.apple.AppKit              	0x00007fff92de9efa _DPSNextEvent + 1067
9   com.apple.AppKit              	0x00007fff92de932a -[NSApplication _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 454
10  com.apple.AppKit              	0x00007fff92ddde84 -[NSApplication run] + 682
11  libqcocoa.dylib               	0x000000010773437f 0x107712000 + 140159
12  QtCore                        	0x00000001045907a1 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 401
13  QtCore                        	0x0000000104594a65 QCoreApplication::exec() + 341
14  opentrack.bin                 	0x000000010345f1eb 0x103455000 + 41451
15  libdyld.dylib                 	0x00007fff9e84b5ad start + 1

Thread 1:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff8eb82efa kevent_qos + 10
1   libdispatch.dylib             	0x00007fff89b9d165 _dispatch_mgr_invoke + 216
2   libdispatch.dylib             	0x00007fff89b9cdcd _dispatch_mgr_thread + 52

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff8eb825e2 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff909a2578 _pthread_wqthread + 1283
2   libsystem_pthread.dylib       	0x00007fff909a0341 start_wqthread + 13

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff8eb825e2 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff909a2578 _pthread_wqthread + 1283
2   libsystem_pthread.dylib       	0x00007fff909a0341 start_wqthread + 13

Thread 4:
0   libsystem_kernel.dylib        	0x00007fff8eb825e2 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff909a2578 _pthread_wqthread + 1283
2   libsystem_pthread.dylib       	0x00007fff909a0341 start_wqthread + 13

Thread 5:
[... 440 lines cut ...]
__LINKEDIT                        97.3M       42 
__TEXT                           178.9M      269 
__UNICODE                          552K        2 
mapped file                        1.1G       26 
shared memory                     16.3M        9 
===========                     =======  ======= 
TOTAL                              1.6G      797 

Model: MacBookPro11,3, BootROM MBP112.0138.B17, 4 processors, Intel Core i7, 2,3 GHz, 16 GB, SMC 2.19f12
Graphics: Intel Iris Pro, Intel Iris Pro, Built-In
Graphics: NVIDIA GeForce GT 750M, NVIDIA GeForce GT 750M, PCIe, 2048 MB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x134), Broadcom BCM43xx 1.0 (7.21.95.175.1a6)
Bluetooth: Version 4.4.4f4 17685, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0512F, 500,28 GB
USB Device: USB 3.0 Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: RAPOO 2.4G Wireless Device
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1

```

sthalik, 2016-05-04:
This isn't infinite recursion but default stack size is not enough for us. Wait.

sthalik, 2016-05-05:
I have an OSX build but there are new issues. This has to wait.

sthalik, 2016-05-22:
@iDevPro please check rc49p1.

iDevPro, 2016-05-22:
```
Process:               opentrack.bin [4630]
Path:                  /Applications/opentrack.app/Contents/MacOS/opentrack.bin
Identifier:            com.github.opentrack
Version:               opentrack-2.3-rc49p1
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           opentrack.bin [4630]
User ID:               501

Date/Time:             2016-05-22 16:46:27.886 +0700
OS Version:            Mac OS X 10.11.5 (15F34)
Report Version:        11
Anonymous UUID:        F85BD5B7-4491-BB32-B224-3CC3F04AD619

Sleep/Wake UUID:       E8149F8E-920C-4309-BB43-62473FB51C72

Time Awake Since Boot: 140000 seconds
Time Since Wake:       62000 seconds

System Integrity Protection: enabled

Crashed Thread:        12  Tracker_PT

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
[4630] stack overflow

Thread 0:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff983e3f72 mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff983e33b3 mach_msg + 55
2   com.apple.CoreFoundation      	0x00007fff8ec011c4 __CFRunLoopServiceMachPort + 212
3   com.apple.CoreFoundation      	0x00007fff8ec0068c __CFRunLoopRun + 1356
4   com.apple.CoreFoundation      	0x00007fff8ebffed8 CFRunLoopRunSpecific + 296
5   com.apple.HIToolbox           	0x00007fff8509a935 RunCurrentEventLoopInMode + 235
6   com.apple.HIToolbox           	0x00007fff8509a76f ReceiveNextEventCommon + 432
7   com.apple.HIToolbox           	0x00007fff8509a5af _BlockUntilNextEventMatchingListInModeWithFilter + 71
8   com.apple.AppKit              	0x00007fff8d01edf6 _DPSNextEvent + 1067
9   com.apple.AppKit              	0x00007fff8d01e226 -[NSApplication _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 454
10  com.apple.AppKit              	0x00007fff8d012d80 -[NSApplication run] + 682
11  libqcocoa.dylib               	0x000000011393437f 0x113912000 + 140159
12  QtCore                        	0x00000001108967a1 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 401
13  QtCore                        	0x000000011089aa65 QCoreApplication::exec() + 341
14  opentrack.bin                 	0x000000010f760fb4 0x10f757000 + 40884
15  libdyld.dylib                 	0x00007fff8595c5ad start + 1

Thread 1:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff983eaefa kevent_qos + 10
1   libdispatch.dylib             	0x00007fff8a52d165 _dispatch_mgr_invoke + 216
2   libdispatch.dylib             	0x00007fff8a52cdcd _dispatch_mgr_thread + 52

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff983ea5e2 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8fc7a578 _pthread_wqthread + 1283
2   libsystem_pthread.dylib       	0x00007fff8fc78341 start_wqthread + 13

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff983ea5e2 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8fc7a578 _pthread_wqthread + 1283
2   libsystem_pthread.dylib       	0x00007fff8fc78341 start_wqthread + 13

Thread 4:
0   libsystem_kernel.dylib        	0x00007fff983ea5e2 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8fc7a578 _pthread_wqthread + 1283
2   libsystem_pthread.dylib       	0x00007fff8fc78341 start_wqthread + 13

Thread 5:: com.apple.NSEventThread
[... 415 lines cut ...]
CG shared images                   432K       14 
CoreAnimation                      108K       18 
CoreUI image data                  960K        9 
CoreUI image file                  320K        5 
Dispatch continuations            16.0M        2 
Kernel Alloc Once                    8K        3 
MALLOC                           108.7M       43 
MALLOC guard page                   32K        7 
Memory Tag 242                      12K        2 
OpenCL                               8K        2 
Process Corpse Info               2048K        2 
STACK GUARD                         76K       20 
Stack                             72.7M       22 
VM_ALLOCATE                         52K        8 
__DATA                            21.8M      272 
__IMAGE                            528K        2 
__LINKEDIT                        97.5M       45 
__TEXT                           179.3M      271 
__UNICODE                          552K        2 
mapped file                        1.1G       27 
shared memory                     16.3M        9 
===========                     =======  ======= 
TOTAL                              1.6G      786 

```

sthalik, 2016-06-16:
@iDevPro run opentrack.sh from a terminal and see if there's any scary warnings. I cannot reproduce any of the problems.

sthalik, 2016-09-21:
OSX isn't supported since there's no maintainer. We need a programmer that can use a Mac daily.

closed 2016-09-21 by sthalik