opentrack/opentrack #351: Using a 'cap' IR model, got strong corelation between yaw and z!!

freds72, 2016-05-01:
Tracking of the model is working fine - got a red marker on the top LED (cap dimension 60/60 and 85mm).
Thing whenever I turn left/right, Z value follows almost 1:1!

Trying to adjust the red marker to align with my actual head center yields very little effect (and appears to be very unstable even after a position reset...).

Where should be located the red marker? Between eyes? Center of head?
There is actually very little information on the PT setup!

Len62, 2016-05-01:
Should be center of head. Have you performed the calibration step? Once calibration is complete the red X should fall somewhere near the center of your head.

sthalik, 2016-05-04:
Hey,

The wiki instructions for calibration should be fine for both caps and clips.

freds72, 2016-05-04:
Auto-calibration goes nuts, "sees" center of head as far as 300 mm away from cap!!
Using manual calibration (0,-60,60), I got better results (program is DCS World).
However, it still have weird artifacts:
- correct center position (looking right):
[image]
- same position, looking slightly down:
[image]

Notice how the center jumped from ok position to completely off (yellow lines gives cap shape, arrow is where I am looking at).

Is my cap model too small for the IR tracker to differentiate between top/left/right??

For information, I am using PT to track a VR "homemade" headset, I can rotate head way beyond "screen" cone. 
What are the "safe" limits for PT head rotation?

sthalik, 2016-05-04:
It not only "jumps" but gives incorrect rotation from the raw data on screenshots.

Enable dynamic pose resolution and increase the timeout to a couple of seconds.

severgun, 2016-05-12:
As I know red cross should be slightly under ears if looking in profile and at center if looking from face. This is your skull turn point

closed 2016-06-16 by sthalik