opentrack/opentrack #352: HTC Vive support

PaulKeeble, 2016-05-09:
Would be really nice to get Vive support similar to the Oculus via SteamVR.

MyrddinE, 2016-05-10:
Second that. There are a lot of games that support TrackIR... it would be very pleasant to play them in my Vive with head tracking inside the Steam cinema, Virtual Desktop, or similar.

sthalik, 2016-05-10:
Can't we simply support all devices the SteamVR library supports?

sthalik, 2016-05-10:
openvr project link: https://github.com/ValveSoftware/openvr

We have here a "multiple CRT" problem. We can either live with that and avoid `std::` memory layout issues _or_ create a subprocess.

There's a version for OSX IA-32 so we should use a subprocess since our OSX build is amd64 only.

Bearnosen, 2016-05-24:
I´m eagerly awaiting this!! Any progress on this front?

Bobbehhh, 2016-05-24:
As am I! I just received my Vive and just assumed OpenTrack already had support for it. Can't wait till it does. Thanks!

sthalik, 2016-05-24:
Hey,

I'm working on support for an eye tracker right now. It's done except for relative view mode. Give it a few days and I'll finish it.

After I include SteamVR, someone needs to test it. It's also unlikely to work the first time.

MyrddinE, 2016-05-24:
I would be happy to test it. I have several games with full TrackIR support but no Vive support.

Bobbehhh, 2016-05-24:
I would also be happy to help test it. I have 16 years of experience in QA testing with video card HW/SW, game testing and general computer product development. Do you have a HTC Vive? Would having one help with getting this to work? if so, I'm sure we could setup some type of funding page (or something less public).

sthalik, 2016-05-24:
I appreciate the help guys. It's gonna be the next feature then.

Don't send me any money, please. We can talk on Skype once I have time, then you get the code.

The later Rift trackers with positional tracking I made just sending random improvements to the person testing it. It worked after few iterations.

Bearnosen, 2016-05-29:
Any progress? :)

IPFR33LY, 2016-05-30:
Lots of people who bought the Vive, having support for it within openTrack would be a huge bonus.

Bearnosen, 2016-06-09:
Any news? :smiley:

sthalik, 2016-06-09:
No, and you're not helping. Having me receive the notification each time an "are we there yet?" message is sent to github doesn't write any code.

Bearnosen, 2016-06-09:
I'm sorry, didn't mean to stress you out!

Bearnosen, 2016-07-30:
Sthalik - I've got no experience of QA testing like Bobbehhh, but I got a Vive and I'd be happy to help any way I can! Would wire you some money if you would accept it for the effort you put in to this project!

sthalik, 2016-07-30:
Hey,

I don't take donations. It's a very important feature but I'm exhausted.

Bearnosen, 2016-07-30:
Oh ok, take your time man! You don't owe us anything so don't overdo it!

sthalik, 2016-07-30:
If you want to donate to opentrack, donate to EFF instead. Thanks.

sthalik, 2016-09-26:
I'd rather support only OSVR. See #294. Is there any device that works with openvr only? It's closed-source despite the confusing name.

sthalik, 2016-11-02:
I'll be supporting this before OSVR.

sthalik, 2016-11-02:
See if this works. https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa

If it doesn't, tell me what's broken.

smarky55, 2016-11-02:
Just tested this with latest steamVR beta
Getting "steamvr compositor error: 103" while tracking

krisp, 2016-11-02:
same result as comment above. Set to steamvr input, click start, says error 103 repeatedly in the console, no video and no tracking.

sthalik, 2016-11-02:
Try this - https://www.dropbox.com/s/rlx4wtbs1sgv17b/opentrack-htc-vive-test_2.7z

This build lies to SteamVR so that it provides the pose data.

smarky55, 2016-11-02:
Seems to be mapping rotation to translation:

- Pitch to tz
- Yaw to ty
- Roll to tx

(This is with default opentrack settings)
also getting "tracker: backlog interval overflow 10005244" in the console about every 15 seconds

sthalik, 2016-11-02:
Thanks. The rotation to translation thing was a typo.

See if this helps with backlog: https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa

If tracking won't work with this version, see if it works after starting a game rendered to the device.

smarky55, 2016-11-02:
Seems to be working well, no backlog overflows. Translational axes are in metres, don't know if you would want to scale. Rotational axes are ok but default mapping is still mixed and switching them in the settings is a bit weird (guessing due to the order the rotations are applied). 
Current profile ini attached (as zip cus github)
[default.zip](https://github.com/opentrack/opentrack/files/567282/default.zip)

sthalik, 2016-11-03:
Does it still work in that build while a game's rendered on the device? In build 4 I don't register as a drawing app to avoid problems as per above.

See if this build works. I'm no longer using the compositor so tracking can be totally broken.

If it actually works see if there's any latency.

I fixed the translation scale and adjusted rotation order. Thanks.

https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa?dl=0

sthalik, 2016-11-03:
Also see if any signs are inverted, thanks.

sthalik, 2016-11-03:
Also see the build `4-alt` that uses a better way of accessing the pose data. It doesn't use the compositor either so it won't interfere with tracking.

The `4-alt` way is better since the pose is less delayed. Please confirm which (or both, or neither) of the builds work.

sthalik, 2016-11-03:
When checking, please confirm inverted status for all six (xyz, and yaw pitch roll). Also see if the rotation order is now correct.

If the `4-alt` build works then we can skip the `4` build.

JamesG3D, 2016-11-03:
The 4-alt seems to work for me; the 4 build didn't work at all as far as i could tell. 

The position (TX, TY, TZ) seem to be all working very well, and seem to pretty much match 1 to 1 with cm as far as I could tell. 👍 

However all the rotation values all change simultaneously when only rotating on one axis. 
For reference, I took some screenshots while only changing the yaw:
http://imgur.com/a/yPF9V
I also recored the tracking data while I was taking those screenshots, using the recored tracking data button under developer in SteamVR settings. Unfortunately I have no idea what format it is:
https://dl.dropboxusercontent.com/u/49076428/yaw-LHR-DA7BD9DE_1478159952_00.zip

When only changing pitch I got these values, in degrees; (yaw, pitch, roll):
pitch -90  (180, -77, 180) - looking down
pitch -45  (180, -99, 180)
pitch 0     (0    , 0   ,    0  ) - straight ahead
pitch 45   (13  ,-37 ,    5  )
pitch 90   (9    ,-72 ,    7  ) - looking straight up
pitch 180 (172,  -2 ,  180) - looking backwards (upside down)
Another tracking file while doing this:
https://dl.dropboxusercontent.com/u/49076428/pitch-LHR-DA7BD9DE_1478159952_01.zip

sthalik, 2016-11-03:
Try all three centering methods please.

sthalik, 2016-11-03:
It's likely different rotation order. We can fix it bug let's eliminate the possibility of wrong center method.

sthalik, 2016-11-03:
We also need axis-angle conversion code in opentrack itself to avoid gimbal lock.

JamesG3D, 2016-11-04:
Changing the centering method seems to have had some effect on the rotation values, but they still all change simultaneously. Relative seems like it might be better? I'm not sure. I made a video of my test if thats any help:
https://dl.dropboxusercontent.com/u/49076428/2016-11-05%2009-06-34.mp4

sthalik, 2016-11-04:
Was the device upside down?

If you put it the other way so roll is close to zero, is it better? I can rotate it upside down after the pose is transmitted if that helps.

danielfullmer, 2016-11-04:
@sthalik @JamesG3D

For me, everything works fine locally about (0,0,0) - (yaw, pitch, roll)

If I keep yaw centered near 0, then pitch and roll work fine as well, through their full range of motion.
However, if I keep just yaw 90 degrees to the left or right, the pitch and roll invert.
As I continue to 180 degrees yaw, it appears that pitch and roll are backwards at this point.

In the attached zip file, I've included images of opentrack with the vive in the forward, 90 left, and 180 backwards positions, and the associated steamvr room overview images.
[opentrack-vive-pictures.zip](https://github.com/opentrack/opentrack/files/572662/opentrack-vive-pictures.zip)
 
I'm using the 4-alt build above. The 4 build didn't work at all for me either. I didn't notice any difference changing the centering method.

JamesG3D, 2016-11-04:
Nah it was the right way up, but yeah it does seem to work a bit better with the headset upside down.

sthalik, 2016-11-05:
These are two things

1) need ypr extraction for steamvr rotation order
2) need to and back from axis-angle conversion to work around gimbal lock

That's gonna take a few days at least. I'm working on translating opentrack right now.

aomas, 2016-12-01:
Wait a minute will we be able to output data to SteamVR? because that would  be amazing.

sthalik, 2016-12-01:
No, only input from steamvr. VR library is more than head tracking, in particular stereo image processing. This is outside the scope of the opentrack project.

wenxiao2012, 2016-12-26:
@sthalik 
I tried  opentrack-2.3.0, the same thing happend like @JamesG3D  shows in his video.
Will this problem solved in next version?

sthalik, 2016-12-26:
I don't know. It takes a lot of unpaid work to add new hardware support. At the moment I'm sort of burnt out from all the effort.

Tobii EyeX and HTC Vive are the highest priority new features.

sthalik, 2016-12-26:
On the other hand you can contribute conversions between axis-angle and rotation matrix representations. We also need to convert to Euler from the particular rotation order HTC vive API uses.

The hardware support is in, it's only these math formulae that are missing.

wenxiao2012, 2016-12-27:
@sthalik
Is there any tools to read the raw output of vive? then we can try to help with the conversions.

sthalik, 2016-12-27:
This is the last line in opentrack for getting the rotation matrix https://github.com/opentrack/opentrack/blob/unstable/tracker-steamvr/steamvr.cpp#L102

sthalik, 2016-12-27:
See if this build works: https://www.dropbox.com/s/j227an9b4w2d8da/opentrack-test-20161226-htc-vive.7z?dl=0

JamesG3D, 2016-12-27:
Awesome, I'll take a look at it when I get back home on the 1st jan.

sthalik, 2016-12-27:
keep in mind to use the link before testing. I already noticed some bugs in the formulae and updated the file in-place.

wenxiao2012, 2016-12-28:
I tried opentrack-test-20161226-htc-vive.7z， still not right.

sthalik, 2016-12-28:
How does it work in that version?

wenxiao2012, 2016-12-28:
[LP_20161228210640.zip](https://github.com/opentrack/opentrack/files/675598/LP_20161228210640.zip)
I put my HMD on a stool，and rotate it. then you can see how the yaw and pitch value changed.

sthalik, 2016-12-29:
I've read that

- matrix is row-major *probably*
- order is YXZ -- yaw, pitch, and roll
- it might be transposed (?)
- code, taken from "G3D Innovation Engine"

<pre>
bool Matrix3::toEulerAnglesXZY (float& rfXAngle, float& rfZAngle,
                                float& rfYAngle) const {
    // rot =  cy*cz          -sz              cz*sy
    //        sx*sy+cx*cy*sz  cx*cz          -cy*sx+cx*sy*sz
    //       -cx*sy+cy*sx*sz  cz*sx           cx*cy+sx*sy*sz

    if ( elt[0][1] < 1.0f ) {
        if ( elt[0][1] > -1.0f ) {
            rfXAngle = (float) G3D::aTan2(elt[2][1], elt[1][1]);
            rfZAngle = (float) asin( -elt[0][1]);
            rfYAngle = (float) G3D::aTan2(elt[0][2], elt[0][0]);
            return true;
        } else {
            // WARNING.  Not unique.  XA - YA = atan2(r20,r22)
            rfXAngle = (float)G3D::aTan2(elt[2][0], elt[2][2]);
            rfZAngle = (float)halfPi();
            rfYAngle = 0.0f;
            return false;
        }
    } else {
        // WARNING.  Not unique.  XA + YA = atan2(-r20,r22)
        rfXAngle = (float)G3D::aTan2( -elt[2][0], elt[2][2]);
        rfZAngle = -(float)halfPi();
        rfYAngle = 0.0f;
        return false;
    }
}
</pre>

Stay tuned.

wenxiao2012, 2016-12-31:
I have read the openvr docs, and some sample projects. the tracking menthod of vive is toooo complex... I cant figure it out. but i still can help to test.

Nich10solo, 2017-02-05:
sthalik, any updates on this issue?  Any way we can help?  Thanks.

sthalik, 2017-02-05:
@Nich10solo you can write the code, given the rotation method is already known (see above).

I'm out of doing serious work for the time being.

subnet-, 2017-03-26:
Hello there, I worked a bit through the code and came up with this working solution.  
Things I'm not sure about:  
- sensitivity (previous code had XYZ * 100)
- roll around 90° (seems to be wonky because of asin() )  

Here's the Code:  

```cpp
void steamvr::data(double* data)
{
    if (vr)
    {
        vr::TrackedDevicePose_t devices[vr::k_unMaxTrackedDeviceCount] = {};

        vr->GetDeviceToAbsoluteTrackingPose(vr::ETrackingUniverseOrigin::TrackingUniverseSeated, 0,
                                            devices, vr::k_unMaxTrackedDeviceCount);

        bool done = false;

        for (unsigned k = 0; k < vr::k_unMaxTrackedDeviceCount; k++)
        {

            if (!devices[k].bPoseIsValid)
                continue;

            if (vr->GetTrackedDeviceClass(k) != vr::ETrackedDeviceClass::TrackedDeviceClass_HMD)
                continue;

            const auto& result = devices[k].mDeviceToAbsoluteTracking;

			// the previous code had the position multiplied by 100
			int sensitivity = 100;

			data[TX] = result.m[0][3] * sensitivity;
			data[TY] = -result.m[1][3] * sensitivity;
			data[TZ] = result.m[2][3] * sensitivity;

			// transformation matrix to euler angles
			if (result.m[0][0] == 1.0f || result.m[0][0] == -1.0f)
			{
				data[Yaw] = atan2f(result.m[0][2], result.m[2][3]);
				data[Pitch] = 0;
				data[Roll] = 0;
			}
			else
			{
				data[Yaw] = atan2f(-result.m[2][0], result.m[0][0]);
				data[Pitch] = atan2(-result.m[1][2], result.m[1][1]);
				data[Roll] = asin(result.m[1][0]);
			}

			static constexpr double r2d = 180 / M_PI;

			data[Yaw] *= r2d;
			data[Pitch] *= r2d;
			data[Roll] *= r2d;

            done = true;
            break;
        }

        if (!done)
            qDebug() << "steamvr: no device with pose found";
    }
}
```

Here's a quick video:
https://youtu.be/LOjYhEOn59w

sthalik, 2017-03-28:
@subnet- that's excessively amazing. Thank you! Please provide a name you'd like to be credited as in the source code, as well as whether you're fine with the ISC license or would like to use a different one.

Roll value is a result of limited domain/codomain of `asin`. You can see an alternative method here, <<https://github.com/opentrack/opentrack/blob/unstable/tracker-pt/ftnoir_tracker_pt.cpp#L201>>
```
beta  = atan2( -R(2,0), sqrt(R(2,1)*R(2,1) + R(2,2)*R(2,2)) );
```

but I can't perform the change myself not knowing the rotation order, and missing the device.

sthalik, 2017-03-28:
In worst-case scenario with have `2(9 choose 2)` possible solutions. That is 72. We could try them all if it's not obvious.

[96 more replies]

closed 2017-09-30 by sthalik