opentrack/opentrack #354: Intel RealSense IR Camera

severgun, 2016-05-12:
Is it possible to use IR camera(640*480 200fps) as regular point tracker? Now it show green screen.

I did not test RealSense tracker through sdk quite well but want to know, is plan B possible?

ph0b, 2016-05-12:
It's not possible as the IR camera isn't directly exposed to the system as a regular camera.
It'd be possible to develop such module by using [librealsense](https://github.com/IntelRealSense/librealsense) to get access to the realsense camera, the IR stream and feed it back to the point tracker.

severgun, 2016-05-12:
SR300 at Device List shows as SR300 RGB and SR300 IR

sthalik, 2016-05-12:
We already have an abstraction over the camera PT. It needs very little logic for a new implementation.

If anyone wants to do this, you can link against the PT module directly. Relevant symbols aren't exported but that's a line or two to change in PT and the build system.

severgun, 2017-02-23:
I see some movement about realsense cams in last issues. Is there any chance to get this feature implemented? I would like to do it by myself, but I'm not a dev. :(