opentrack/opentrack #356: Octopus having artifacts at specific settings

MathijsG, 2016-05-13:
On specific settings here, being 90 degrees pitch in-game, and the last position before 20 degrees in physical, my octopus is tripping all over.

I've made a video of it: https://www.youtube.com/watch?v=9y9I5DdqR9s

As far as I see it doesn't affect the output, however this effect seems weird to me.

sthalik, 2016-05-13:
Good catch on reproducing it. I'll take a look.

sthalik, 2016-05-13:
@MathijsG please confirm fixed https://www.dropbox.com/s/rsj78idet70clqt/opentrack-pt-v15.7z

MathijsG, 2016-05-14:
Still having the issue? is your upload the correct one? It's not stating some like V15.

[image]

sthalik, 2016-05-14:
@MathijsG please try https://www.dropbox.com/s/krt76r4zp2owkc5/opentrack-pt-v16.7z

It works for me with yaw, pitch, and raw all enabled. The archive naming is arbitrary.

MathijsG, 2016-05-14:
Working here now! Great work man!

Was this problem introduced by something recently? I can't remember having this weirdness earlier on.

sthalik, 2016-05-14:
@MathijsG probably not. But even if, the numerical code has to remain stable in light of recent changes. There's a new flag for vectorization now but the thing I fear here is crashes on just some CPUs.

closed 2016-05-14 by sthalik

sthalik, 2016-05-14:
@MathijsG and thanks for the amount of work you put into submitting and discussing issues during the whole 2.3 development process.

commit 17a05a38 (2016-05-14): pose-widget: don't promote to double

Committer shouldn't push so fast to get a chance to rebase.

Closes #356