opentrack/opentrack #357: semantic versioning problems

sthalik, 2016-05-13:
The versioning style for the project is messed up with the "release candidate" things that aren't really rc's.

How about we go for 2.3.1, 2.3.2 etc. with new features and 2.3.2p2 (where `p` stands for "patchlevel") with only bugfixes?

I'd like to use a sensible version scheme without version number bloat like Chrome or Firefox. Is there a better alternative?

sthalik, 2016-05-13:
Pinging @ph0b @dbaarda

MathijsG, 2016-05-13:
Why not use the last digit for bug fixes instead of extra patchlevel semantics.

So every fix will raise the 2.3.X, instead of 2.3.2PX

I don't see the benefit of the patchlevel raise, instead of the version as a whole. It makes it confusing for people about running which version imo.

sthalik, 2016-05-13:
@MathijsG but if we increment the middle value with each new feature, legacy branches look like `branch/2` not like `branch/2.4`.

sthalik, 2016-05-13:
To make it clearer - new feature would rise to `2.4`, `2.5` and so forth. I'd like to keep the two first values for actual sweeping changes, not less than every quarter.

MathijsG, 2016-05-13:
What about:

Version 1.0.0
Bug fixed: 1.0.1
New feature(s): 1.1.0
Huge changes: 2.0.0

sthalik, 2016-05-13:
Then a legacy branch will be `branch/2` or `branch/3`. We don't have the manpower to support feature releases as legacy branches. Fair enough. The next rc will be the last rc for 2.3.

There was a compile flag change that could crash things. Let's wait for more feedback before finalizing the change.

And yes, what you propose is typical semantic versioning unlike my scheme.

ph0b, 2016-05-14:
I prefer typical semantic versioning without patchelevels.
The last value can be for bug fixes and minor changes too. This way we can keep the first couple of values for changes that are more relevant to the users (new trackers/filters, etc).

dbaarda, 2016-05-15:
The approach used for libs is <major>.<minor>.<micro> where for bug fixes
you bump the micro version, for backwards compatible feature extensions you
bump the minor version, and for non-backwards compatible changes you bump
the major version.

This way anything that works with version 2.4.0 should work with any
version 2.4+, like 2.4.3, 2.6.2, 2.12.17, etc, but not with 2.3.12, or
3.8.9.

It's easier to classify changes to libraries than applications into major,
minor, micro version bumps, because its easier to identify the effects of
API changes than user interfaces, but sometimes apps have external
interfaces for things like plugins that you can version for.

sthalik, 2016-05-22:
We're not preserving the ABI however. The opentrack-api module can have the ABI changed at any time. Other than that I agree with relatively significant functionality being the minor.

sthalik, 2016-06-16:
Fixed. We're close to a release.

sthalik, 2016-08-15:
I'm just scared of a point release.

MathijsG, 2016-11-29:
Congrats on your 2.3.0. release Stanisław! Good job :-).

I will continue to add reports and will fine tune the Dutch translation.

sthalik, 2016-11-29:
🐱  👍 

Closing fixed.

closed 2016-11-29 by sthalik