opentrack/opentrack #358: RealSense+F200 -> Opentrack crashing when starting tracking

J-Fricke, 2016-05-14:
Just installed 2.3 rc21p11 Updated Drivers to latest for the F200 and installed 2016 R1 SDK and I get a crash every time I start tracking. It says I need the SDK runtime even though it's installed.

ph0b, 2016-05-14:
Sorry about that.
It works fine with my setup, let's try to see what could cause the crash:
Can you reinstall the SDK runtime using the installer which is in opentrack?
You say you've got the SDK, are the head tracking sample working fine with it?

Is the crash instantaneous? 
is the camera lighting up before the crash?
Do you get any error number, or more details about the crash?
can you check if the result is the same with [2.3-rc21p7](https://github.com/opentrack/opentrack/releases/tag/opentrack-2.3-rc21p7) ?

J-Fricke, 2016-05-15:
I have just uninstalled all drivers and open track. Then I installed again in this order.

1. f200_1.4.27.41944
- Important Windows Updates
- OpenTrack 2.3 rc21p11
- RealSense SDK (included with OpenTrack)
- Created new config
- Start tracker

- First attempt crash open track
- Second attempt 'status code -3' error dialogue box with Options: OK, Install Runtime, X

sthalik, 2016-05-15:
@ph0b can you make a p7 or newer build that works on your system? It works with the GNU toolchain (see `mingw-w64.cmake`) and theoretically with MSVC++ 2015.

ph0b, 2016-05-17:
-3 means tracking isn't available from the SDK. That means it somehow breaks during the first attempt that crashes.

@sthalik your builds and mines all work on my machine. 

@J-Fricke thanks a lot for the additional information. I need to reproduce the error you're facing but i don't have the F200 with me this week. Can you precise what OS you're using? Is the crash happening right after you start tracking, or there is some delay? does the preview windows become black before the crash? are the camera green and red lights lighting up before the crash? can you put this [dll](https://my.syncplicity.com/share/jvmv0llelkknhvd/opentrack-tracker-rs-impl) inside opentrack folder and try again?

J-Fricke, 2016-05-18:
win 8.1
Right after
black
no lights (can get green light and video with PointTracker)

J-Fricke, 2016-05-18:
Does not work with that dll either.

sthalik, 2016-05-22:
@ph0b try SDK_CONSOLE_DEBUG to true and qDebug() output.

ph0b, 2016-05-24:
Sorry for the delay, I was traveling and just got back home.
I've tried to reproduce the error on a win 8.1 system with an F200, and unfortunately for you, it just work on this machine too.

Since the init of the SDK works (you're getting an error message from the SDK module at the second try), I suspect the crash comes from the SDK directly. The root cause could be a faulty USB3.0 connection with your camera or another app using the camera directly at the same time. But I'm not 100% sure yet.

@J-Fricke Are you able to use the F200 with any other application?
There is a new release of the SDK coming up very very soon. I'll update it in opentrack and do a build with more debug and safeguards.

J-Fricke, 2016-05-31:
@ph0b I use it all the time for the microphone, it even works displaying standard video with the SDK and even a couple of other tracking types in Opentrack it will turn on and give output. It seems directly related to the point tracker with the SDK.

sthalik, 2016-06-01:
@ph0b you might want to send a debug build with gdb.exe bundled. Just saying.

ph0b, 2016-06-01:
@J-Fricke I was asking more about apps that would use the SDK. You may try this face tracking sample from it: https://my.syncplicity.com/share/zma0pi3gmxwadu9/DF_FaceTracking

The new RealSense SDK release is out and I've made a debug build with it for you: https://my.syncplicity.com/share/oov0ajlojpnj4xw/opentrack-debug

If it's still crashing, run gdb/gdb.bat at the same time than opentrack.
Start tracking, and once it has crashed, launch the command 'thread apply all bt' inside gdb.
This will generate a log file next to gdb.bat after a minute, with all the stack trace information, please upload it here and tell us what was displayed inside opentrack debug console before the crash (tracking has started, etc).