opentrack/opentrack #359: [UX] Clarify which is rotation and which is translation

MathijsG, 2016-05-20:
When I see the filter options
[image]

I always need to google for which was which again. Maybe clarify the axes instead of using the abstract terms 'translation' and 'rotation'?

For foreigners they may be less common.

sthalik, 2016-05-20:
Hey,

Even for native speakers "translation" is hard to understand. The term is too technical. For example NASA calls avoiding debris by the ISS "translating", i.e. moving the ISS by thrusters. So it's inappropriate especially for the already opaque filter options.

Looks like that right now, are we good?

[image]

MathijsG, 2016-05-20:
Maybe add this for the sake of simplicity, something like this?

Rotation filtering (axes: pitch yaw, roll)

Position filtering (axes: X, Y, Z)

sthalik, 2016-05-20:
Are we good?

[image]

MathijsG, 2016-05-20:
Yeah I think that will clear some things up! :)

commit 53288d02 (2016-05-20): filter/accela: use less technical language in the UI

To name position tracking "translation" per se is unusual outside of
geometry and hard sciences in general. Clear things up.

Reported-by: @MathijsG
Closes #359
closed 2016-05-20 by sthalik