opentrack/opentrack #361: FreePie Android - Wrong axis calculation?

FlintEastwood, 2016-05-23:
If turning the phone in yaw- direction, everything is fine.
But turning in roll or pitch direction, the squid turns on three axis instead of one.
I tried to swap roll and pitch in FreePie input settings, but in roll direction the squid turns again on three axis.
Its hard to describe, but output data freaks out compared to input data.

Device is a Sony Z3 Compact with the FreePie IMU.
OpenTrack is 2.3 rc49p1

sthalik, 2016-05-24:
Put the device in a neutral position like you track it. Go to freepie tracker settings and add offsets to rotation so that it's not more than 90 on each.

The detailed explanation is that Euler angles have multiple solutions describing the same rotation. And there is no mathematical way to work around it.

FlintEastwood, 2016-05-26:
> The detailed explanation is that Euler angles have multiple solutions describing the same rotation. And there is no mathematical way to work around it.

Oh sorry, I didn't know that since I'm new to this. I did some Googling now and it is all logical to me now. It is the Gimbal-Lock problem, right?
Thank you for your help. :)

sthalik, 2016-05-27:
There's an another issue at hand here. Mapping is done at the end. If mapping was done at the beginning, these problems would never arise. But moving mapping affects everything including even filtering.

closed 2016-06-21 by sthalik