opentrack/opentrack #363: oculus rift cv1

Piffx1, 2016-06-05:
Hi all,
does actually opentrack works with oculus rift cv1?
 greetings
Piff

sthalik, 2016-06-05:
Not yet. I've got my hands full with work, opentrack, and personal.

Kitten just turned six weeks and it's one hell of a time investment.

sthalik, 2016-06-09:
Some progress. Soon-ish ovr SDK 1.4.0 for testing.

Piffx1, 2016-06-20:
oh great ! thx for the info

bizzclaw, 2016-07-13:
If you need help with testing, just let me know. Just got my CV-1 in yesterday.

sthalik, 2016-07-14:
Some stuff is done for cv1 but it's still not certain it'll work.

Piffx1, 2016-07-16:
That would be great! what's your nickname in oculus

sthalik, 2016-07-19:
see #339

closed 2016-07-19 by sthalik

Tbsc74, 2016-08-24:
Just wondering what the progress is for opentrack being compatible with Oculus Rift CV-1. Really want to play Arma 3.

Piffx1, 2016-08-24:
i buying only for arma 3 the oculus rift and a gtx1080 XD