opentrack/opentrack #366: Main LED in PointTracker 1.1 switching around

archeria1, 2016-06-07:
When I move my head up the red main LED always switches around. I made sure to enter the exact positions of the 3 LEDs on my clip on, every LED is always visible and there are never more than 3 dots visible. How can I force him to keep the main LED the main LED? It's not like it disappears.

MathijsG, 2016-06-07:
Can you make a screenshot? Did you calibrate your clip or hat?

Len62, 2016-06-08:
Sounds like you did not do the calibration steps. Here's a link to a video on calibrating. Once done you should have 3 green and 1 red dots visible.
https://github.com/opentrack/opentrack/wiki/Quick-Start-Guide-(WIP)

sthalik, 2016-06-08:
@archeria1 enable dynamic pose resolution.

archeria1, 2016-06-10:
@sthalik Thanks, that seems to have done the trick. It still bugged out sometimes but I just adjusted the curves so I don't have to turn until that critical point.

closed 2016-06-10 by archeria1