opentrack/opentrack #367: PT tracker Problem with pitch and roll

katsM, 2016-06-09:
Hello

I can't get PT tracker to work. 
Everything is calibrated (red dot well placed) and settings seem OK.
But pitch affects only roll...

What did I miss?
Everything was fine with 2.3Rc18. I deleted old profiles before testing this.

sthalik, 2016-06-10:
Model coordinates wrong.

katsM, 2016-06-10:
> Model coordinates wrong.
You mean clip dimensions?
I tested with the default ones, and the following:
        26
                       43
                       57
75

        27
                       43
                       62
75
After several tests with camera offset and its position on my screen, I can now pitch, but roll is affected whatever I do... And when I roll pitch is strongly affected also. 
Thanks for your help

sthalik, 2016-06-10:
You can't use arbitrary coordinates that were made for another clip. Use a ruler and get exact measurements.

katsM, 2016-06-10:
> You can't use arbitrary coordinates that were made for another clip. Use a ruler and get exact measurements.

I did, with same results.

Len62, 2016-06-10:
Do you have any dead space set-up on the mapping page? And which filter are you using? I find I can't keep my head perfectly still and adding deadspace and/or adjusting the accela filter can help.

katsM, 2016-06-10:
> Do you have any dead space set-up on the mapping page? And which filter are you using? I find I can't keep my head perfectly still and adding deadspace and/or adjusting the accela filter can help.

Hi
I use Accela filter and I've smoothed my curves in mapping screen to avoid some unwanted little movement.
Is there an order for the settings? I did:
1-Model settings, 2-calibration, 3-camera offset
Since I use the Natural Point Track Clip pro, should I use the clip tab for model or custom?
For now I'm totally unable to obtain a roll without pitch and vice versa. Yaw is OK.

Len62, 2016-06-10:
Use the clip tab and double check the measurements. Try setting the camera offset to zero, and then re-calibrate. In the accela filter settings you can adjust the sensitivity/deadzone for both the rotation and translation axes. When you pitch up/down any deviation of head movement will affect the translation, likewise for when you roll, except the deviation will affect the rotation.

sthalik, 2016-06-11:
Camera offset will only get rid of e.g. rolled clip's yaw bleeding into pitch and roll. Before doing camera offset disable "center at startup" and don't center. Align with "raw data" and watch the values' signs. It should be close to (0,0,0) and if adding camera offset increases the values, the sign is probably wrong.

Like @Len62 said, until you achieve proper tracking don't touch calibration. If you correct model coordinates, recalibrate.

sthalik, 2016-06-11:
There's a change where camera offset degrees are of the same sign as raw data degrees. Unfortunately it's a breaking change, and users need to flip the sign. See per 18f375918765baf7427a54c27863cf5e0fa19d5f that only roll offset is affected

Good news is that I mounted my HOTAS and will actually use my own software. This hasn't happened in more than a year.

sthalik, 2016-06-11:
It wouldn't be mine without the duct tape.

[image]

sniperwolfpk5, 2016-06-11:
haha nice book shelf

MathijsG, 2016-06-11:
Lol you're like my friend who also uses his empty case as a bookshelf (http://www.techeblog.com/index.php/tech-gadget/pc-case-bookshelf-mod?&cuid=1a773be5b0c493f236aa784db85734da)

What games are you playing with your own software? :-)

Len62, 2016-06-11:
Nice way to re-purpose old desktops, bookstand + flight controls!

sthalik, 2016-06-11:
@MathijsG Falcon BMS 4.33u1 with modded DB and X-Plane. For the latter, see my friend:

[image]

It's his first experience with flight. A good friend of mine.

sthalik, 2016-06-16:
@katsM try first to center the camera on yaw physically and apply camera roll offset in settings -> camera, no other offsets. See if that makes a difference.

@katsM sorry for hijacking your thread.

katsM, 2016-06-16:
> @katsM try first to center the camera on yaw physically and apply camera roll offset in settings -> camera, no other offsets. See if that makes a difference.

Will try this. My camera is positionned on the top of my screen. Should the model be centered on the video view or it doesn't matter?

 > @katsM sorry for hijacking your thread.

Np it was fun anyway :)

sthalik, 2016-06-16:
You don't need to center the model physically. Just ensure that raw data yaw points close to zero. You don't need to calibrate either, unless you have a valid calibration already.

Values are according to raw data passed through mapping. From your description there's no gimbal lock. If you have sudden spikes, make sure pitch and roll go less than 180 after mapping.

Also try this fresh build if you want. There _should_ be no regressions. https://www.dropbox.com/s/qkn52lyqww4jexe/opentrack-mouse-test-20160615.7z

katsM, 2016-06-16:
Thanks a lot

Will try to test this tonight after work

katsM, 2016-06-16:
Followed your advices, it's way better now.
I placed the camera to obtain a raw yaw at 0 and then made all the settings.
It's not perfect (roll still affects pitch and vice versa but just a little now), but I think I've got some clues to get the job done.
Thank you very much everyone!

sthalik, 2016-06-19:
@katsM I'm glad it works better for you. The changes are in 2.3.0 rc 3.

closed 2016-06-19 by sthalik