opentrack/opentrack #369: RFC: axis interconnect specification

sthalik, 2016-06-11:
Is it useful to have axis interconnect specified for each axis? For instance, pitch adding downward movement or yawing to -180/180 zooming for looking backward in car sims.

We can have for each axis six tabs using our normal mapping interface. Only that more complex stuff like requiring two axis to have a certain value in unison is too involved to implement without allowing for user code.

Do you want it?

sh

MathijsG, 2016-06-11:
It would definitely can come in handy indeed for looking back like you said.

MathijsG, 2016-06-15:
Does this also solve things like this? https://github.com/opentrack/opentrack/issues/292

That you can either increase or decrease axis interconnection? Or am I just mistaken? :-)

sthalik, 2016-06-15:
A good UI including "move toward center" hasn't been proposed, so no.