opentrack/opentrack #370: mouse emulation proplem

MrKossmo, 2016-06-11:
Hello! I use opentrack with my smartphone as VR head mount display. But there is tracking problem. I put on my headset, sit down 360 spining chair, and play first person shooters/survival (even not supports headtracking). At 180+/- angle opentrack does 360 turn around, and it makes me crazy spoiling immersoin to a game. Please help me to solve that problem!

MrKossmo, 2016-06-12:
Mapping properties? I don't understand what kind of curve you mean. I should use near to 1:1 linear ratio (real head rotation is the same to a virtual). In my headset I reach it at 180 to 120 (see pic). Horizontal line (lower than 180) make angle behind straight view (180+/-) more wide, and it looks like deathzone. It's not allows continious 360 view.

https://pp.vk.me/c633121/v633121060/2ea53/qf8yAt2ZIRI.jpg

The problem i see is that my phone data overcome -179 to +179 through shorter path (through 180). But Opentrack output data overcome the same point throught longer way (full rapidly flip from -179 to zero, and from zero to +179). Opentrack doesn't goes cycle. It' looks awful when i walk ingame labyrinth, move straight on course +179/-179 in real and and my virtual head does that crazy flip

MrKossmo, 2016-06-12:
Intugame VR app mouse tracker works as i want. Permanent smartphone rotation (360 and more) transfers to permanent pointer movement to the same direction. No flips. But this application is very bad at streaming video and it doesn't works in background mode (if i want to use only tracker).

sthalik, 2016-06-15:
@MrKossmo please see if this helps any https://www.dropbox.com/s/qkn52lyqww4jexe/opentrack-mouse-test-20160615.7z

sthalik, 2016-06-20:
@littlewolfpaint @MrKossmo can you confirm 2.3.0 rc 4 works?

sthalik, 2016-07-10:
I believe it works now. Closing due to lack of response.

closed 2016-07-10 by sthalik