opentrack/opentrack #375: rc49p5 crashes when I click "start"

Emton, 2016-06-23:
Crashes when I click start
[image]

Last version I was using was from a few weeks ago with out any problem, I installed using the exe, did I forget something or did my computer setup change somehow to cause this?

Emton, 2016-06-23:
Uh ok I did a clean uninstall, then I installed the portable version and I think it works.

Emton, 2016-06-23:
Oh I spoke too soon it's actually crashing as soon as it detects the Aruco shape
[image]

sthalik, 2016-06-23:
Send a camera frame with your marker in it.

kblomster, 2016-06-23:
I'm having the same issue, it works fine as long as the aruco marker isn't in frame but when it enters, boom. It's a regression, 2.3 rc49p2 doesn't have the issue. I've attached frame with the marker in it.

[image]

sthalik, 2016-06-24:
@kblomster see if pre6 works for you, thanks.

kblomster, 2016-06-24:
Installed rc49p6, but I still have the same issue. Starting the tracking without the marker in frame works fine, but as soon as the marker is detected (red outline appears around it), the application crashes. I don't have Visual Studio on this computer, but I could probably install it and try to get you a stack trace or something - I have no experience with OpenCV or anything though so I doubt I'll be of much use attempting to debug the problem.

Would it help if I tried to narrow down exactly which version it broke in?

sthalik, 2016-06-24:
@kblomster version with link-time optimization has no visible symbols or debug info. They can't be embedded. It probably broke in pre3 with the compiler change enabling working `-fipa-pta` on Windows. And `-fipa-pta` also miscompiles Rift SDK 025. I'll give you a test version in a few.

sthalik, 2016-06-24:
@kblomster see if https://www.dropbox.com/s/ecab4japfylhok2/TEST_opentrack-2.3-rc49p6_2-win32-portable.7z?dl=0 works, thanks

kblomster, 2016-06-24:
Tried it (shows version as rc49p6-dirty) but nope, still same issue.

sthalik, 2016-06-25:
@kblomster here's a debug build. Visual studio wouldn't show you exact frame information, but use gdb. In particular, use this build of gdb. Before running the process, input

`catch throw`
`break break __cxa_throw`

If the latter doesn't work, just continue debugging.

After it crashes, run

`thread apply all bt`

and send the output.

This is only to ensure that it works without any optimizations. If this version works, the compiler needs rolling back to 5.3.

sthalik, 2016-06-25:
@kblomster the links

https://www.dropbox.com/s/4fnelemxcyakt10/opentrack-debug-20160625.7z?dl=0
https://www.dropbox.com/s/l4c8csuude1am8s/opentrack-gdb-20160625.7z?dl=0

kblomster, 2016-06-25:
I'm gone over the weekend and will be able to try it on Monday at the earliest. Sorry for the delay.

sthalik, 2016-06-25:
@kblomster no, worries. You're doing a lot for the project this way actually.

sthalik, 2016-06-27:
@kblomster here's a build after backing off the compiler change - https://www.dropbox.com/s/1n5in6puv4uh14e/opentrack-test-20160627.7z

sthalik, 2016-06-27:
Rift 0.2.5 is no longer miscompiled. It's likely that aruco will now work.

kblomster, 2016-06-27:
Hm. Tried test-20160627, still same issue.

Tried the debug version, which also crashes - here's the debugger output and backtrace:

```
C:\opentrack-debug>c:\gdb\bin\gdb.exe opentrack.exe
GNU gdb (GDB) 7.10.50.20160130-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "mingw32".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from opentrack.exe...done.
(gdb) break throw
Function "throw" not defined.
Make breakpoint pending on future shared library load? (y or [n]) n
(gdb) catch throw
Catchpoint 1 (throw)
(gdb) break __cxa_throw
Function "__cxa_throw" not defined.
Make breakpoint pending on future shared library load? (y or [n]) y
Breakpoint 2 (__cxa_throw) pending.
(gdb) run
Starting program: C:\opentrack-debug\opentrack.exe
[New Thread 8656.0x119c]
options: registering stream operators
Loading "libopentrack-filter-accela.dll"
Loading "libopentrack-filter-ewma.dll"
Loading "libopentrack-tracker-aruco.dll"
Loading "libopentrack-tracker-freepie-udp.dll"
Loading "libopentrack-tracker-hatire.dll"
Loading "libopentrack-tracker-joystick.dll"
Loading "libopentrack-tracker-pt.dll"
Loading "libopentrack-tracker-rs.dll"
Loading "libopentrack-tracker-udp.dll"
Loading "libopentrack-proto-fgfs.dll"
Loading "libopentrack-proto-freetrack.dll"
Loading "libopentrack-proto-simconnect.dll"
Loading "libopentrack-proto-udp.dll"
Loading "libopentrack-proto-win32-mouse.dll"
bundle + "opentrack-ui"
joy list refresh
[New Thread 8656.0x10ac]
[New Thread 8656.0x1e9c]
[New Thread 8656.0x13d0]
[New Thread 8656.0xa84]
add joy "{6B73DF90-8C04-11E2-8001-444553540000}"
got joy "{6B73DF90-8C04-11E2-8001-444553540000}"
[New Thread 8656.0x1a28]
[New Thread 8656.0x2118]
add receiver 56134560
joy list refresh
bundle + "proto-freetrack"
bundle + "aruco-tracker"
bundle + "Accela"
[New Thread 8656.0x29ac]
add receiver 132898736
joy list refresh
[New Thread 8656.0x1dfc]
[New Thread 8656.0xa04]
[New Thread 8656.0x283c]
[Thread 8656.0x283c exited with code 0]
[... 128 lines cut ...]
#2  0x771a17c3 in ntdll!RtlpQueryDefaultUILanguage () from C:\Windows\SysWOW64\ntdll.dll
#3  0x76c5338a in KERNEL32!BaseThreadInitThunk () from C:\Windows\syswow64\kernel32.dll
#4  0x77189902 in ntdll!RtlInitializeExceptionChain () from C:\Windows\SysWOW64\ntdll.dll
#5  0x771898d5 in ntdll!RtlInitializeExceptionChain () from C:\Windows\SysWOW64\ntdll.dll
#6  0x00000000 in ?? ()

Thread 1 (Thread 8656.0x119c):
#0  0x7717016d in ntdll!ZwWaitForMultipleObjects () from C:\Windows\SysWOW64\ntdll.dll
#1  0x7717016d in ntdll!ZwWaitForMultipleObjects () from C:\Windows\SysWOW64\ntdll.dll
#2  0x76bf171a in WaitForMultipleObjectsEx () from C:\Windows\syswow64\KernelBase.dll
#3  0x00000002 in ?? ()
#4  0x0028dde0 in ?? ()
#5  0x76c51a08 in WaitForMultipleObjectsEx () from C:\Windows\syswow64\kernel32.dll
#6  0x76590892 in USER32!PeekMessageW () from C:\Windows\syswow64\user32.dll
#7  0x66b3e58e in ZN21QEventDispatcherWin3213processEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE ()
   from C:\opentrack-debug\Qt5Core.dll
#8  0x0028df84 in ?? ()
#9  0x6df25597 in qwindows!qt_plugin_instance () from C:\opentrack-debug\platforms\qwindows.dll
#10 0x0028fd00 in ?? ()
---Type <return> to continue, or q <return> to quit---
#11 0x66aeb095 in ZN10QEventLoop4execE6QFlagsINS_17ProcessEventsFlagEE () from C:\opentrack-debug\Qt5Core.dll
#12 0x66af2fe8 in ZN16QCoreApplication4execEv () from C:\opentrack-debug\Qt5Core.dll
#13 0x00402326 in main (argc=1, argv=0xc8b560) at ../gui/main.cpp:64
(gdb)
```

sthalik, 2016-06-27:
Hey @kblomster,

There's a new logic error after simplifying aruco. Try this build - https://www.dropbox.com/s/o0ghg4x89v6prq1/opentrack-test-20160627_2.7z?dl=0

You can run it under the debugger to get the error message. It's only that the backtrace won't be visible.

sthalik, 2016-06-27:
If there are still errors please record a camera video for me to reproduce under manycam.

kblomster, 2016-06-27:
Well, *something's* changed. Running 20160627_2 under gdb worked for a few seconds, but then crashed again. gdb shows the message "warning: Invalid parameter passed to C runtime function." twice when it crashes and then a bunch of threads exiting with code 3, nothing else of interest as far as I can see (let me know if you want the full output).

Will record camera video shortly.

kblomster, 2016-06-27:
Will [this clip](https://drive.google.com/open?id=0B8bCDRcq9BVeNDNfTmlGV3JVeHc) do? Let me know if you want something longer or with a different codec/file format or anything (this is h.264, 640x480 @ 60fps).

sthalik, 2016-06-27:
@kblomster great clip, thanks. It no longer crashes on that clip. See https://www.dropbox.com/s/ntluwx14f7hs9b8/opentrack-test-20160627_3.7z

Other than the newly-introduced bugs in 2.3 release process, there was the matter of never checking height to be greater than zero. Since forever.

sthalik, 2016-06-27:
What the hell, I didn't close the issue

sthalik, 2016-06-27:
Good news.

Windows Subsystem for Linux has a fully-working Ubuntu Xenial environment. It's also possible to run the openssh server as a Windows service. Now I no longer have to build Qt, opentrack et al. on mingw or Cygwin. WSL provides an almost complete Linux environment with binary compatibiliy.

kblomster, 2016-06-27:
I can confirm that I can't get 20160627_3 to crash - it seems to work just fine. Thanks!

sthalik, 2016-06-27:
@kblomster thanks for working to get it fixed.

commit 10c47b15 (2016-06-27): tracker/aruco: fix #375

The height was zero on the test video. The ROI check only saw if width
is at least zero. Check for both to be greater than one.

Video provided by: @kblomster
Issue: #375

Also, fix minor issues:

- nix vars that can be const static in function scope
- don't call solvepnp twice where obj_points shift will do
- don't do bounds checking on vector elt access
- respect sprintf varargs type size; change to snprintf for no reason
- fix clamp-to-image logic
- set proper alpha for fps estimation
closed 2016-06-27 by sthalik