opentrack/opentrack #376: Opentrack with iRacing

CRBlocker, 2016-06-23:
I just started getting into iRacing and I wanted to play around with Opentrack. Unfortunately, it doesn't want to work with it. 
The only related piece of information I could find regarding both iRacing and Opentrack is found here: [https://github.com/opentrack/opentrack/issues/246](url). The two suggestions there are to double check UAC settings and leaving iRacing "Look Left" and "Look Right" hotkeys as 'Unassigned.' Both are set as such, but tracking still doesn't register in game. 
I'm using the smartphone Opentrack solution, version 2.3_rc5_d with FreePIE UDP receiver as the Tracker and freetrack 2.0 Enhanced as the Protocol, and Accela as the Filter. 
iRacing executable used is iRacingSim64DX11.exe. I've also tried it with iRacingSimDX11.exe, the 32-bit version.
Running Windows 7 x64, GeForce GTX 660.
Although iRacing is on an entirely separate drive than the Windows installation, I have tried to set Admin privileges to the EXE, but launching the game as normal from the browser does not allow it.
Opentrack is fantastic. I've been really enjoying it in other games like Elite Dangerous and American Truck Simulator ever since I discovered it a few days ago, but it'd be great to get some help for iRacing. Thanks in advance.

sthalik, 2016-06-23:
send a packed executable.

CRBlocker, 2016-07-08:
Sorry, I'm not entirely sure what you're asking for. Do you want the iRacingSim64DX11.exe packed in Rar/zip? 
As an update, I've noticed that the tracking does actually work within iRacing, but only on the first launch of the game after a computer reboot. If I were to, for example, do a practice session with it working after reboot and switch to a real race it wouldn't work within the real race because it's a separate game launch.
I thought this might be happening because of some hanging process. I ended every process I could on both the PC and the Android phone (since that's also part of the setup), and also tried keeping Opentrack up between the game launches, as well as opening Opentrack fresh with the other game launch. I also tried turning off Steam overlay and DX11 in game. Nothing worked.
Thanks for responding and trying to help.

sthalik, 2016-07-09:
What would happen if you launched another game using the freetrack protocol between two iRacing sessions? It's not a solution, but I'm trying to find the piece that's broken.

CRBlocker, 2016-07-09:
Seems I was quick to think a reboot resolved it. It was working with Elite Dangerous, but didn't work with iRacing afterwards.
I rebooted and opened three games in a row, in this order: iRacing, Elite Dangerous, American Truck Simulator. All didn't work. All times I opened Opentrack and connected after the game was well launched. 
I updated to the latest 2.3-rc49p7 version and made sure I kept the same Input, Output, and Filter settings. The gui shows solid tracking in the new version. I also noticed an Nvidia graphical update, so I throw that on. I then scanned and cleaned the registry with IOBit Systemcare, even though it didn't list anything related to the issue. I also set Steam to run in admin mode since all three games launch with it. 
I rebooted and opened the same three games, in the same order, and all three still didn't work.
I'm not sure how all three games have now stopped working. I used to only have problems with iRacing, but something must be completely broken now. Other than Steam, I don't use any overlays or things that otherwise tie into the game from a third-party. The last few attempts with iRacing I've had the steam overlay set off from a setting within the game's browser configurator. I think I've mentioned everything to note.
Do you have a recommended order of operations when you use Opentrack with games? Are there common issues to watch for, in general, that I should checklist?
I'm sorry it's now become a much bigger problem than before. I'll need to do some more searching around tomorrow since it's late for me. 
Until then, I do appreciate any thoughts you might have. Thanks again.

sthalik, 2016-07-09:
> Do you have a recommended order of operations when you use Opentrack with games? Are there common issues to watch for, in general, that I should checklist?

Launch opentrack with freetrack protocol. Then go to registry HKCU -> Software -> NATURALPOINT -> NaturalPoint -> NPClient Location -> Path. The value of the "Path" string should point to the opentrack install.

Copy the "Path" key's value and paste it into the Run dialog (win + r). It should point to where opentrack is, and NPClient.dll as well as NPClient64.dll should be there.

As for ED, opentrack needs to be installed on the same partition as ED.

CRBlocker, 2016-07-09:
The NaturalPoint "Path" field was blank, so I added 'D:/Applications/opentrack/', which is the install directory, and also where Run opened an explorer window to reveal both the NPClient.dll and NPClient64.dll. My D: drive contains all three games, Opentrack, and is without a Windows installation.

Without a reboot, I decided to test.
I opened ED, then OT, didn't work.
OT, then ATS, worked, closed/opened OT, worked.
OT, then iR, worked, closed/opened iR, worked, closed/opened OT, worked.
OT, then ED, worked, closed/opened OT, worked.
iR, then OT, worked, closed/opened OT, worked.
ED, then OT, worked.

I think I did enough testing for now to say that the registry Path was the issue. Not sure why, but maybe it needed some time before it kicked in, which might have been the problem with the first ED attempt. 
Something to note is that ATS mouse and keyboard look controls don't work now, even without OT running. I don't see myself playing without OT, so it's not a problem, personally. But I figured I'd mention it. If that sort of thing is something you might consider an issue down the line, I'd be glad to help test it out for you.

It seems the problem has been resolved, but I realize the fickle nature of things. Still, I can't thank you enough for helping me through it. This has become a required bit of software that changed the game for me, so to speak, so I'd be happy to donate and contribute what and how I can. 

Thanks so much.

sthalik, 2016-07-10:
@CRBlocker there's an option to clear the Path field when stopping opentrack, in the freetrack protocol dialog. Keep it disabled. Also ensure that you have "using NPClient" in freetrack protocol dialog.

samkatakouzinos, 2016-08-21:
@CRBlocker I use iRacing / Opentrack / DelanClip. 
I use PointTracker 1.1 / freetrack 2.0 Enhanced / Accela.

Headtracking seems to work best with opentrack ready before iRacing is loaded. 

I advise the use of the 'Game Detection' feature within opentrack, located in Options. 
You provide a list of game executables and which opentrack profile to use for each game. 

More importantly is the option to 'Start profiles from game executables name in this list'. 
When ticked this will start AND stop opentrack between game plays. 

This feature may better enhance your experience with opentrack considering the number of games you want to use it with.

sthalik, 2016-08-21:
You're saying that opentrack should be started before the game. Why is that? Game detection is a personal preference.

sthalik, 2016-09-21:
I think we're good here.

closed 2016-09-21 by sthalik