opentrack/opentrack #378: Aruco resetting yaw on rotation.

Ajay-Ramachandran, 2016-06-30:
When I rotate to make a different acuro marker visible, everything is reset as if I was using the old marker and my head was rotated 180°.

sthalik, 2016-06-30:
In aruco wiki https://github.com/opentrack/opentrack/wiki/Aruco-tracker

see:

"Please keep in mind that marker has to be waved at the camera at a neutral pitch angle of at least 15-25 degrees -- since it's planar, solver gets confused when pose approaches identity. When mounting it on a hat, ensure a relatively oblique angle that keeps the raw pitch value positive at all times. Otherwise, sudden spikes in pitch value ensue."

closed 2016-06-30 by sthalik

Ajay-Ramachandran, 2016-06-30:
@sthalik The pitch was fine, it is just that when one marker goes offscreen, it does not assume the other marker to be a different marker

sthalik, 2016-06-30:
@Ajay-Ramachandran it's not meant to. There's presently no support for multiple markers at differing orientations,

Ajay-Ramachandran, 2016-06-30:
Oh, ok. Thanks