opentrack/opentrack #379: Mouse as input

andrewpros, 2016-07-05:
Hi.

I surprised that this option is not available and no one is asking for this.

Why there isn't a mouse input?

This way you could play any vr game and use mouse as head movement emulator, could work great with riftcat when you gyroscope is not good, have some bugs or you just want something better. (or just don't have gyroscope)

Any plans for this, looks simple enough to add?

sthalik, 2016-07-05:
> Why there isn't a mouse input?

No reason to prove a negative. 

I don't find mouse input very useful. Perhaps you can find a reason justifying programmer time for implementing that.

andrewpros, 2016-07-05:
But i already have described the reason, to emulate head movement, the most important thing, up/down/left/right by mouse. It shouldn't be that hard.

If this don't justify programmer time then probably joystick input shouldn't be there either, but it is and it is doing exactly what i described, you can use it to emulate head movement and play a game without even using any expensive hmd.

But due to x360/xone controller being used already by many vr games to control them using is as a head movement emulation is not viable, hence, emulating it by mouse is the way to go.

I would gladly make it myself, but i don't have experience in hmd's head movement data, don't know the math to translate mouse movement into the right values for games.

So someone experienced with this, like some opentrack developer, could make it much faster than i and probably it isn't something that hard.

Well i could try, there is already freePIE UDP receiver, so i could somehow use it and translate mouse to hmd data, but the problem remains the same, don't know the math, would take me long and it is another software to run.

So i would be great if it would be just build into opentrack and simple to use.

sthalik, 2016-07-05:
Hey,

Joystick emulation is useful for EDtracker and potentially other head-tracking devices exposing themselves as USB HID. In fact HID was expected to be the right thing to use for this purpose.

Most simulation software comes preequipped with mouse view control on its own. That's my experience. Even if not, the ability to control yaw and pitch with a mouse isn't especially enthralling. I'll give it consideration if it's often requested.

sh

andrewpros, 2016-07-05:
Ok, fair enough, just looking a way to play those tiny games without headset.

sthalik, 2016-07-06:
You need a headset or an HMD. Otherwise there's no immersion. With head tracking it's lots.

Closing. Like I said I'll reconsider if more people ask.

closed 2016-07-06 by sthalik

andrewpros, 2016-07-06:
No immersion, like any other non vr games and people still playing them.

Sartoni, 2016-08-27:
It would be great to use the Air mouse , for example in the head to simulate the movements and would be much more accurate I am layman in programming but it would be great if someone prontificasse to help