opentrack/opentrack #380: Bug in point_tracker.cpp

DaMichel, 2016-07-06:
I just found a little issue here https://github.com/opentrack/opentrack/blob/unstable/tracker-pt/point_tracker.cpp#L49

In this part
```
for (unsigned i = 0; i<points.size(); ++i)
    d_order[i] = d_vals[i].second;
```
`points.size()` can be more than the size of d_order which amounts to `PointModel::N_POINTS=3 `. It happened to me and it resulted in a stack corruption warning in the debugger. I guess the warning was legit ...

sthalik, 2016-07-06:
Thanks, great catch!

closed 2016-07-06 by sthalik

DaMichel, 2016-07-07:
Wait a sec. `points.size()` can be less that 3, can't it? So I think it might be better to take `min(points.size(), PointModel::N_POINTS)`.

sthalik, 2016-07-07:
@DaMichel no. See that `PointTracker::track(...)` is called only if `success` is true in the thread.

<pre>
const bool success = points.size() >= PointModel::N_POINTS;

if (success)
{
    point_tracker.track(points, PointModel(s), fx, s.dynamic_pose, s.init_phase_timeout);
    ever_success = true;
}
</pre>

There are no other code paths that go there.

sthalik, 2016-07-07:
Also see `std::sort` usage in `PointExtractor::extract_points(...)`. We're using the three most circular points.

DaMichel, 2016-07-07:
Ok. Great. Nevermind then.

commit 7a3569f8 (2016-07-06): tracker/pt: fix out-of-bounds write

The input array has exactly 3 points. The vector has potentially more
points, sorted by circularity.

Spotted-by: @DaMichel
Closes #380