opentrack/opentrack #381: Compiling under M$ VC++ 2015 x64

DaMichel, 2016-07-07:
Hello again. I was working to compile opentrack without any unix tools and I found something odd about the DLL export declarations of some classes. For instance, we have
```
class OPENTRACK_COMPAT_EXPORT Timer
```
The linker spits out errors for this class. About unresolved symbols to be precise.
```
Creating library E:/Eigene Dateien/Programmierung/opentrack_build/tracker-pt/Debug/opentrack-tracker-pt.lib and object E:/Eigene Dateien/Programmierung/opentrack_build/tracker-pt/Debug/opentrack-tracker-pt.exp
21>ftnoir_tracker_pt.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: __cdecl Timer::Timer(void)" (__imp_??0Timer@@QEAA@XZ) referenced in function "public: __cdecl Tracker_PT::Tracker_PT(void)" (??0Tracker_PT@@QEAA@XZ)
21>point_tracker.obj : error LNK2001: unresolved external symbol "__declspec(dllimport) public: __cdecl Timer::Timer(void)" (__imp_??0Timer@@QEAA@XZ)
21>ftnoir_tracker_pt.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: void __cdecl Timer::start(void)" (__imp_?start@Timer@@QEAAXXZ) referenced in function "protected: virtual void __cdecl Tracker_PT::run(void)" (?run@Tracker_PT@@MEAAXXZ)
21>point_tracker.obj : error LNK2001: unresolved external symbol "__declspec(dllimport) public: void __cdecl Timer::start(void)" (__imp_?start@Timer@@QEAAXXZ)
21>ftnoir_tracker_pt.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: __int64 __cdecl Timer::elapsed(void)const " (__imp_?elapsed@Timer@@QEBA_JXZ) referenced in function "protected: virtual void __cdecl Tracker_PT::run(void)" (?run@Tracker_PT@@MEAAXXZ)
21>point_tracker.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: long __cdecl Timer::elapsed_ms(void)const " (__imp_?elapsed_ms@Timer@@QEBAJXZ) referenced in function "public: void __cdecl PointTracker::track(class std::vector<class cv::Vec<float,2>,class std::allocator<class cv::Vec<float,2> > > const &,class PointModel const &,float,bool,int)" (?track@PointTracker@@QEAAXAEBV?$vector@V?$Vec@M$01@cv@@V?$allocator@V?$Vec@M$01@cv@@@std@@@std@@AEBVPointModel@@M_NH@Z)
21>E:\Eigene Dateien\Programmierung\opentrack_build\tracker-pt\Debug\opentrack-tracker-pt.dll : fatal error LNK1120: 4 unresolved externals
```
I think it expects the members to be defined in a compilation unit of the DLL, i.e. in opentrack-compat.dll, rather than directly in the header. Removal of `OPENTRACK_COMPAT_EXPORT` fixes the issue for me.

I uploaded my changes in a new branch here: https://github.com/DaMichel/opentrack/commit/590fbe15bcf5510582d9cc1f81fa2ffdadbd8d8e

Moreover, I needed a few more changes to make it work. Firstly, I disabled the tarball build target. Actually I probably did it wrongly. Should have checked for `_MSC_VER` instead of `_WIN32`. Secondly, I made building of the Intel RS tracker optional because I happen to make an x64 build for which you didn't supply import libs (only x32). And I didn't want to bother with it.

I would like to make a pull request. But I figure I show you first. Note that I didn't test my changes on Linux.

sthalik, 2016-07-07:
@DaMichel your changes make sense except for minor details in line comments at the commit you linked to. Please rebase. I'd also like to know what's wrong with the realsense tracker.

sthalik, 2016-07-07:
@DaMichel in cmake you can do `if(MSVC)` and it works for .sln and nmake builds like that.

As for word size we can compile another RS for 64-bit.

DaMichel, 2016-07-07:
Will take care of it later on.

Regarding RS, the issue is just that I don't have a 64-bit version of `opentrack-tracker-rs-impl.dll`. A 64-bit build would be appreciated. Based on what I read (http://www.mingw.org/wiki/Interoperability_of_Libraries_Created_by_Different_Compiler_Brands), it should link successfully even if you compile the library with MinGW. Because `C` interface.

sthalik, 2016-07-07:
@DaMichel pull from repo and try again.

sthalik, 2016-07-07:
Pinging @ph0b since he may want a say in all that.

sthalik, 2016-07-08:
@DaMichel but you don't need to run "make tarball". It works only with msys/cygwin in the path anyway.

DaMichel, 2016-07-08:
Doh! This is true. I need to turn off building of the project in VC's build configuration.

It actually kinda worked with sh.exe from git. However then it was missing `zip` ...

So might as well leave the `if (~MSVC)` in there, okay?

sthalik, 2016-07-08:
Ok, fine. Can you see if latest vjoy changes make proto/vjoy link?

DaMichel, 2016-07-08:
Well I didn't even try to build vjoy support earlier. Now I tried. There are two issues:
Firstly, I downloaded vjoy SDK. Directory structure is different from what you expect. It is
```
SDK\lib\vJoyInterface.dll
SDK\lib\vJoyInterface.lib
SDK\lib\amd\vJoyInterface.lib
SDK\lib\amd\vJoyInterface.dll
```
Secondly, `VJOY_API` was defined two times. The following works for me:
```
#ifndef _MSC_VER
#define VJOY_API extern "C"
VJOY_API BOOL __stdcall VJoy_Initialize(PCHAR name, PCHAR serial);
VJOY_API VOID __stdcall VJoy_Shutdown();
VJOY_API BOOL __stdcall VJoy_UpdateJoyState(int id, PJOYSTICK_STATE pJoyState);
#else
#define VJOY_API extern "C" __declspec(dllimport)
VJOY_API BOOL __stdcall VJoy_Initialize(PCHAR name, PCHAR serial);
VJOY_API VOID __stdcall VJoy_Shutdown();
VJOY_API BOOL __stdcall VJoy_UpdateJoyState(int id, PJOYSTICK_STATE pJoyState);
#endif
```

By the way, my changes to the export declarations are apparently not harmful to the Linux build. I build everything except parts that depend on opencv successfully. 

Who would have thought, the most recent Mint distro ships only the outdated version 2.4 of opencv ...

sthalik, 2016-07-08:
@DaMichel clone opentrack-depends for vjoy. The .dll files are copied from the `dll/` subdirectory. The SDK you want is http://headsoft.com.au/download/pc/VJoySDK.zip

sthalik, 2016-07-08:
As for C linkage in MSVC case you're right.

DaMichel, 2016-07-08:
Interestingly the M$ link does not like the `extern "C"` statement.
Looking at https://github.com/opentrack/opentrack-depends/blob/master/vjoy/VJoy.h#L2, I guess they are really exported as C++ function. With name mangling

sthalik, 2016-07-08:
@DaMichel okay. Send me a pull request if it's still broken.

DaMichel, 2016-07-08:
Thanks! Finally works!

closed 2016-07-10 by sthalik

sthalik, 2016-07-10:
@DaMichel want to get commit rights?

DaMichel, 2016-07-10:
Sure. I would appreciate it.

sthalik, 2016-07-10:
@DaMichel send me an email or better yet, add `sthalik` on skype. Please don't change core stuff without discussing it first.

sthalik, 2016-07-10:
@DaMichel an team invitation has been sent.