opentrack/opentrack #382: Support for PlanetSide2?

petsild, 2016-07-07:
Hi,
is there any way to Opentrack could work in the future in PlanetSide2?
https://www.planetside2.com/home/youtube
I tried to emulate the mouse and in the menu I am successful, but in the plane there is no reaction. Here are the complicated instructions on how it could work.

https://www.youtube.com/watch?v=WlMF5LkvUK4

or

https://www.youtube.com/watch?v=9RhehulgnkQ

Does anybody have any experience that would help under this game to work OT?
Thank you all very much for your reply. Pet

sthalik, 2016-07-08:
@petsild vjoy would work.

petsild, 2016-07-08:
I tried Vjoy version:

https://sourceforge.net/projects/vjoystick/files/latest/download?source=files

and

http://www.headsoft.com.au/index.php?category=vjoy

In system it is added a virtual device, but OT does not work vjoy ends the message.

[image]

sthalik, 2016-07-08:
Headsoft should work.

sthalik, 2016-07-08:
See device manager if there's a reboot required. Try running as administrator or as non-administrator.

petsild, 2016-07-08:
Vjoy works well... unfortunately the administrator didn't help.

[image]

sthalik, 2016-07-08:
Try rebooting. Try both devices. Windows might sort them incorrectly. vjoy exposes two joysticks. Also see if all is fine in device manager.

I haven't used vjoy in a while but it always worked and there were no changes.

sthalik, 2016-07-08:
Just to be clear, if it's really broken I'll fix it.

petsild, 2016-07-08:
Device manager is error free.
Thought I paste installed files vjoy to the OT path: E:\opentrack\contrib\vjoy it didn't help.

sthalik, 2016-07-08:
Okay.

Don't copy-paste the .dll. Just in case if you pasted any stuff into VJoy install dir then reinstall it.

The vjoy SDK and runtime were updated and that's probably it. Try this build: https://www.dropbox.com/s/fw80gm178v2nddl/opentrack-vjoy-20160708.7z

sthalik, 2016-07-08:
Also if you're using an older vjoy version, update from headsoft's site.

sthalik, 2016-07-08:
If the protocol won't load, rename to "vjoy.dll".

petsild, 2016-07-08:
Now in the menu is missing completely.
- the renaming of the VJoy on vjoy.dll menu doesn't help.

sthalik, 2016-07-08:
This one works for me right now, with the driver: https://www.dropbox.com/s/wio1mhsyg2507w0/opentrack-vjoy-20160708_2.7z

petsild, 2016-07-08:
Still it behaves the same as the p7.
He has to show some settings after confirming OK?

sthalik, 2016-07-08:
Confirming what? Select the protocol and it'll transmit.

Reinstall the driver. The options dialog has the URL.

If it doesn't help, go to device manager, select "uninstall" on vjoy and tick the "remove driver" checkbox. The latter part is important. Then reboot and reinstall the driver from the dialog's site.

petsild, 2016-07-08:
With the driver in the link I'm working from the beginning of the experiments. Has to open some new window after the OK confirmation?

[image]

When you start OT in the background and an attempt to map in the menu Vjoy no reaction to the movement of the octopus.

[image]

sthalik, 2016-07-08:
The dialog buttons aren't functional. In fact I'll change them to a single button saying "ok".

Restart opentrack after driver install just to be sure.

petsild, 2016-07-08:
I manually uninstalled vjoy in the device manager. After rebooting and re-installing no change.

[image]

sthalik, 2016-07-08:
What's your Windows version?

petsild, 2016-07-08:
Windows 7 64bit

sthalik, 2016-07-10:
@petsild  can you see it if it works in a pristine Windows VM? There's the test tracker that generates nonsensical data.

sthalik, 2016-07-10:
@petsild alternatively try some other software doing output to vjoy.

petsild, 2016-07-10:
After countless restarts I gave up.
In addition, I found that the PS2 is so sensitive to the periférii after adding the virtual joystick ignores all the actual device, then you cannot map the analog axis.
I really appreciate your support :-D OpenTrack belongs to the most popular programs that I use.
Where to send some money for your great work?

sthalik, 2016-07-10:
@petsild It's a hobby, I don't intend to take any money. But I appreciate that you like the software.

closed 2016-07-10 by sthalik