opentrack/opentrack #385: GRID Autosport don't work anymore since: opentrack 2.3.0 release candidate 8

andregm3, 2016-07-16:
Please check why GRID Autosport headtracking stopped work on this release.

Freetrack.

Len62, 2016-07-16:
Did you re-select the input/output filters as mentioned in the release notes?

andregm3, 2016-07-17:
How to do that ?

Len62, 2016-07-17:
In #1 and #2, note they are labled input and output in the current versions.
https://camo.githubusercontent.com/caa19afd3574a8409badbf55f0353c5fb8b39f4c/68747470733a2f2f7777772e64726f70626f782e636f6d2f732f366f653870377a637679747532356a2f6d61696e73637265656e2e706e673f646c3d31

andregm3, 2016-07-17:
lol sure yes i did, i use UDP Receive and Freetrack, but Trinus VR dont work with that anymore, just head tracker app from playstore and p8 don't work with GRID p7 works

Len62, 2016-07-17:
It was just a guess on my part. Not familiar with the game or UDP. Maybe someone else will chime in.

andregm3, 2016-07-17:
thanks for your attention

sthalik, 2016-07-17:
I'll check with the actual game.

vikingsail, 2016-07-18:
I second this. RC8 doesn't work on UDP+Freetrack 2.0 with ARMA 3 anymore. 
Interestingly, if I put the protocol on "FreeTrack" mode it works even thou everything is messed up. But on "TrackIR" mode nothing happens in ARMA, while Mr.PinkyOctopus shows movement in OpenTrack.
I guess something is broken with the TrackIR protocol on this RC.

sthalik, 2016-07-18:
I see the bug. Installation directory was repeated so it was like `/program files/program files/...`. I'm recalling rc8.

sthalik, 2016-07-18:
@vikingsail @Len62 @andregm3 any issues anymore in rc9? I tested all involved modules.

andregm3, 2016-07-18:
Trimus VR os not working too, Free-Track output from Trinus to UDP input in Open-Track

sthalik, 2016-07-18:
rc9?

andregm3, 2016-07-18:
oh new release, will try right now

sthalik, 2016-07-18:
Oh and start tracking just once before starting the game since it has to update the path in registry.

andregm3, 2016-07-18:
trinus is not moving the octupus

sthalik, 2016-07-18:
This is different. The issue was freetrack protocol not reading the .csv.

andregm3, 2016-07-18:
GRID is back !

sthalik, 2016-07-18:
@andregm3 please be more specific.

andregm3, 2016-07-18:
GRID and Project Cars are working now ! on RC9

but Trinus VR still don't make Octopos move on Open Track anymore.

sthalik, 2016-07-18:
The UDP tracker had no changes really. It can be a firewall or network issue.

andregm3, 2016-07-18:
i have https://play.google.com/store/apps/details?id=com.frozenbacon.headtracker
and that app work, would it be a Trinus VR issue ?

sthalik, 2016-07-18:
Temporarily disable Windows firewall and see if it works. It's really outside the scope of the project to help with issues like that.

andregm3, 2016-07-18:
thanks mate, but i never use windows firewall, the mouse emulation on Trinus VR works, just the sensor to Open Track that don't, i tested in 2 PC one here and other at my job.

vikingsail, 2016-07-18:
RC9 works perfect! Thanks for the good work!

closed 2016-07-18 by sthalik