opentrack/opentrack #386: Further to 'Pitch jumping on yaw #112'

demonesque, 2016-07-16:
Hi there,

Not sure if this is still an open issue but I have the same problem as post 'Pitch jumping on yaw #112' from earlier this year and the Dynamic Pose resolution does not solve this one yet. My cap 3 point system is a bit of a different geometry to most peoples as you will see in the image below with the front 2 points close to my head so I will leave it up to you whether you want to pursue this one or not for the new build. 

[image]

I am using opentrack-2.3-rc49p7-win32-setup and getting the same results as previously reported in that when the lower points yaw and cross over the line of the top point the head center jumps to random locations. I am however able to use version (2.2 Stable) very successfully with only a jump at full left yaw and am happy to continue with that if this is too far out of left-field. Let me know if you would like the data supplied in case 'Pitch jumping on yaw #122' if you want to take this further.

On another topic, I am looking for a Donate site for Opentrack. Have not found one yet......

Thanks.

sthalik, 2016-07-17:
Are the diodes very close to one another when crossing? Compare "very" by their diameter.

This is only a hobby project. Donate to the EFF instead :+1:

sthalik, 2016-07-17:
@demonesque this is unrelated as in #112 user didn't use dynamic pose originally.

demonesque, 2016-07-17:
Hi Sthalik. 

Thanks for getting back. The point distance is about the same as shown in the image. I have made you a video that shows this as well as all the current settings I could think of. Download from here:
https://www.sendthisfile.com/R1inwAnU014AUxcpJXhKE9Nm

I have tried toggling many of the settings that I did not know exactly what they do, and other ones I did, but none have made a difference yet. I will be away for the next week but if there is more testing you want I will do it when I get back. 

(The EFF that you refer to for donations being the Electronic Frontier Foundation? Is that the right one?
Hell of a 'hobby' project dude. Love your work. Much Kudos. Version 2.2 running beautifully in DCS World. Much smoother and steady than the old FreeTrack that it replaces here.)

demonesque, 2016-07-17:
Sorry, did not mean to Close this yet!

MathijsG, 2016-07-17:
Is this the same problem as described here?

https://github.com/opentrack/opentrack/issues/289

sthalik, 2016-07-17:
@demonesque 

That's the right EFF.

Set dynamic pose to about 1000 milliseconds. Other than that, it should work perfectly fine. Send your .ini or just paste it here.

sthalik, 2016-07-17:
@MathijsG no, look at the video. It's as if he got dynamic pose resolution turned off.

demonesque, 2016-07-19:
Hi Sthalik,

I have already tried dynamic pose settings of 1000 5000 10 100 and 1 but the results are still the same. I will send the .ini file when I get back at the end of the week. It is somewhat like the issue 289 but it resets itself when I recenter my head ( or just past centre as shown in the video) . Strange that v2.2 works fine with my rig. 
One thought is that the resolution of my old eye toy camera is only 640x480 could of that do it?

sthalik, 2016-07-19:
@demonesque send a raw camera feed video (compressed) and model dimensions.

demonesque, 2016-07-23:
Hi Sthalik,

.ini file and raw video attached. Thanks.

Looking at the raw video I am wondering if it is actually just my LED points that are too directional? What do you think? Opentrack (and free track) can track them OK at greater rotations than the rotation that they fade out in the video..... (That being said though, O/T v2.2 works fine with my cap setup )

[dcs.zip](https://github.com/opentrack/opentrack/files/379979/dcs.zip)

Thanks Again.

demonesque, 2016-07-26:
Hi Sthalik,

It looks like you might have found something. I will give it a try with the next build. Thanks.

sthalik, 2016-07-26:
Try this one first https://www.dropbox.com/s/s5hexqt98a8vf0k/opentrack-test-20160726.7z

demonesque, 2016-07-29:
Hi Stanisław,

Yes that seems to have done it. I did have to make a new .ini file but once I had done that it seems to be mostly behaving with a dynamic pose of 1000ms.

It is still possible to occasionally get the jammed upside down octopus at extreme head rotations to the left and it does not reset when the head is returned to center, but I think those are at rotations beyond what normal headtracker use would be.

Much kudos dude. Love your work.

MathijsG, 2016-08-01:
Those weird flippings when looking too far left is exactly the thing I also encounter. Only with a hat tracking device. My clip doesn't have this issue.

sthalik, 2016-08-04:
There was a bug and dynamic pose wasn't working at all. That build has a fix.

sthalik, 2016-08-04:
I'm also trying out max distance between old and new frame's points for dynamic pose. It should fix the rest of the problems.

sthalik, 2016-08-05:
Fixed in rc10. Thanks for the report.

closed 2016-08-05 by sthalik

demonesque, 2016-08-06:
Hi Stanisław,

Yes, From what I saw of the code changes you were making I guessed that the dynamic Pose might have been broken.  Thanks for looking into this for me. Love your work.