opentrack/opentrack #387: Trinus VR not working with Open Track anymore

andregm3, 2016-07-18:
Trinus VR is not working, i've tryied all sensors modes and it lost its connection with UDP from Open Track, i don't know if this is a Trinus VR Side issue or Open Track.

I've already disabled firewall

There is another android sensor UDP sender in Android called Head Tracker, that still works: https://play.google.com/store/apps/details?id=com.frozenbacon.headtracker

sthalik, 2016-07-19:
@andregm3 the UDP protocol format hasn't changed at all.

khan213, 2016-08-02:
Hi I had this problem too and I just delete  documents>opentrack-2.3 folder. I start opentrack and set mapping again. 
I hope to help. 
Cheers.

sthalik, 2016-09-21:
rc11 got totally broken UDP input by accident. rc12 is unbroken again. I've tested it before release and opentrack receives data from a Python program that uses the right format.

closed 2016-09-21 by sthalik