opentrack/opentrack #388: [bug] Profile stays on tracker type of previously loaded profile

MathijsG, 2016-07-18:
Reproduce:

Try to change input to something else.

Load other profile.

Boom, the other profile also got switched to this changed input.

sthalik, 2016-07-18:
I can't reproduce in p9.

sthalik, 2016-07-18:
Is your hard drive overloaded a lot?

MathijsG, 2016-07-18:
No, SSD :-). I tried again, but still same problem.

And it defaulted by Aruco in my profile.

sthalik, 2016-07-19:
Does it matter how long you wait between changing a module and switching profiles?

If not, you likely have no write permissions for the first profile. Try changing it with notepad++ and it won't ave.

MathijsG, 2016-07-19:
I can still save everything and change everything. I will try with everything uninstalled and clean.

sthalik, 2016-08-04:
It had a manual save but now the policy is to save implicitly.

Once you select a tracker, does it save normally again?

MathijsG, 2016-08-04:
No I needed to manually force a different change to have the profile saved. Or manually browse another input source and then change it back to the desired one.

sthalik, 2016-08-04:
I understand now. There was a change moving input/output/filter selection to `modules` subgroup in the .ini file. It has to do with that we're saving whole .ini subgroups at once.

As to why it was empty, no idea. It defaults to Accela/freetrack/Aruco for me. Does it still occur to you?

MathijsG, 2016-08-31:
I think it's okay for now. If something acts weird again I will report in again.

closed 2016-08-31 by MathijsG

sthalik, 2016-08-31:
For now I'm busy and can't release rc12 due to the need to deal with fallout. But it's practically ready, sitting there.

MathijsG, 2016-08-31:
You mean the game? Or are you having some nuclear problems over there in Poland? ;-)

sthalik, 2016-08-31:
@MathijsG if I release opentrack-2.3.0-rc12 now I'll have to deal with all the new issues.

But see https://github.com/opentrack/opentrack/issues?q=is%3Aissue+is%3Aopen+label%3Anext-build-2.3.0-rc12

Wish some programmer would take care of the fallout instead of me. I know this is a big thing to ask for. @DaMichel, how do you feel?

sthalik, 2016-08-31:
Of course there's more taken care of than that label.

MathijsG, 2016-08-31:
Would it be helpful if I actively try some issues where the original poster stopped replying whether your fix worked or not? So we can either close that case or further improve the fix. I want to support somehow, but my code knowledge is not that high and my math skills not at all that I can help significantly in programming matters.

MathijsG, 2016-08-31:
Somehow there is still weirdness when opening profiles or something. I tried the new testbuild you made because of the toggle while held bug (which is now fixed though, THANKS!), and now every profile I open gets their curves reset :(. I have all regular rights in my documents folder. Can there be something else? What if settings change, or things get saved differently? Will it just break previously made profiles?

sthalik, 2016-08-31:
Splines are now set using the settings api and I haven't written a migration function yet.

MathijsG, 2016-08-31:
Ah, thanks. So you wouldn't recommend using your testbuilds for regular use I suppose? I often replaced my current opentrack build with the test build. I will not do that again :-P.

sthalik, 2016-08-31:
You can use splines alright except for the need to set them up again just once.

I recommend dev builds if only you can report known bugs and have a bit of patience. For a regular end-user who wants a working product, probably not.

Using a very old build however - solves nothing. An issue won't get fixed unless it's reported.

sthalik, 2016-08-31:
Also I know that control points are unreadable in the config, right now.