opentrack/opentrack #389: Point extraction treshold has very small sweet spot.

MathijsG, 2016-07-18:
Since a few versions the sweet spot of the point extraction treshold is very low.

Too far to the left and you get tons of dots. Too far to the right and it doesn't detect some of the leds anymore.

I never encountered having more visible dots than just three with this 'automatic treshold' checkbox checked. Is this changed or something? I liked the previous behaviour then.

Too few detection:
**
[image]
**

Too much detection:
[image]

Len62, 2016-07-18:
Try setting the min and max sizes to numbers a little below and a little above the reported pixel size.
In the tracking window if your pixel size is say 6 try setting the min/max to 3 and 9 and see what happens. I haven't noticed any change on my set-up.

sthalik, 2016-07-19:
@MathijsG the slider range got increased to a radius of ten total max. We could go for six max radius but will this be enough?

MathijsG, 2016-07-19:
But is it new that it will also show more dots than 3? Because previously I could move it wherever but it would never discover more than my 3 LEDs. Now it also shows dozens of other dots when I move it slightly further.

sthalik, 2016-07-19:
@MathijsG few versions ago the slider range was completely unrealistic. Either too low or too high, it was never in touch with LED sizes.

sthalik, 2016-07-19:
@MathijsG please see build from #391 if the new point extractor makes things better or worse.

MathijsG, 2016-07-19:
Will try later today. What about this approach: only detect the three brightest dots and ignore all the other dots? I thought this is how the function used to be :-). It effectively did that in my setup earlier. When I unchecked the box I always saw too many dots, but when I checked I always saw just 3.

sthalik, 2016-07-19:
@MathijsG there's a bug with wrong image used. I'm recalling p9... :(

FlyingCircus-, 2016-07-25:
@MathijsG As @Len62  suggested, I believe your min size could be the cause of this. It might be that this parameter was reset with the las installation but clearly considering 1 pixel as minimum size seems too low. When I was testing opentrack a few versions ago I noticed that just brightness noise would cause the appearance of multiple 1-2 pixel dots, setting the minimum to 3 pixels discarded all those fake dots. It's reasonable to think that the same could be happening to you.

FlyingCircus-, 2016-07-25:
Oh, I see now we have a nice setting to automatically set the pixel size. @sthalik says something is wrong with it, I would uncheck the "automatic set points size" and try with your own values until that's fixed

sthalik, 2016-07-25:
Can you find something wrong with this unpublished build? https://www.dropbox.com/s/0tl725glfrzpv7b/opentrack-test-20160724.7z

The "sweet spot" is ok now and there's an experimental camera offset translation option.

FlyingCircus-, 2016-07-25:
@sthalik If I activate the "auto threshold" with that unpublished build I got no detection at all. If I uncheck it the min and max sizes work as supposed to

sthalik, 2016-07-25:
@FlyingCircus- use the sliders. They actually work.

FlyingCircus-, 2016-07-25:
@sthalik Sorry, my mistake. I had to put the slider close to its maximum value. I've yet to understand what the checkbox is doing since I still could alter the dot size values and it had an effect on the detection.

sthalik, 2016-07-25:
@FlyingCircus- it toggles automatic thresholding. Then, the slider takes on a different function.

FlyingCircus-, 2016-07-25:
@sthalik I'm curious then about the advantages of using the checkbox since the user still has to manually set 3 parameters (slider, min and max) to suit his needs

MathijsG, 2016-08-04:
@FlyingCircus- see this thread for more info: https://github.com/opentrack/opentrack/issues/355 I agree that it still seems quite complex.

sthalik, 2016-08-04:
@FlyingCircus- it ignores lighting conditions, only caring about points as the brightest image parts, knowing the pixel amount.

sthalik, 2016-09-21:
I think this isn't true anymore for either auto threshold or manual threshold.

closed 2016-09-21 by sthalik