opentrack/opentrack #390: Mouselook locked if i'm not using opentrack

diegooCF, 2016-07-18:
Hi everyone, this is my first post on github and it is to request help :(. I had been using opentrack for a few months and now i can't use my cellphone because i had some problems with him, but the problem is when i want play a game without using opentrack. I can't move the camera with mouse even the opentrack it's closed, i can't solve it, i tried many ways but i can get it.
I have understood that OpenTrack install a "virtual device," but if this is true, I do not know where it is.

Thank you in advance for your help. And sorry for my English if not understood correctly.   :-)

sthalik, 2016-07-19:
@diegooCF opentrack doesn't install a device driver. opentrack puts a path to a .dll in the registry and a game/sim loads that .dll provided as part of opentrack.

I assume you're using the freetrack protocol. Go to protocol settings using the `...` button and select the `disable on program exit` checkbox. From now on, you need to start opentrack first before running any games/sims. If/once you get used to running opentrack with any sims, you can disable the option and the registry path will persist. That option clears the path once opentrack is stopped.

sthalik, 2016-07-19:
The .dll doesn't recognize that opentrack has quit and no pose data is to be sent. I'd like to change it at any moment but Arma's anticheat whitelisted our .dll and I'd like to avoid any political maneuvering.

diegooCF, 2016-07-19:
@sthalik Thanks for you help. Yes, i'm using the freetrack protocol but i don't have the option that you said. I attach a screen so you can see.:

[image]

Thank you again

WF000, 2016-07-20:
What version of Opentrack is that? (it looks older) , 
Newer >2.2  versions have that option right under 'Locate Dll'.
Please update.
Regards

sthalik, 2016-07-20:
@diegooCF rc7 is fine.

diegooCF, 2016-07-20:
@sthalik Thank you very much. I was not aware of a new version. Thank you for solving my problem. By the way, great job!

Regards :-)

closed 2016-07-20 by diegooCF