opentrack/opentrack #391: Accidental interruption of the positioning of the last P9

petsild, 2016-07-18:
Hi,
unfortunately in P9 after a couple of fully functional rcxx an appears accidental stopping the positioning of the head.
Octopus after the suspension moves and shows the data. It helps to just turn off and again turn on the Ot and the game Rise of Flight.

sthalik, 2016-07-19:
Which tracker is it?

sthalik, 2016-07-19:
And protocol too, sorry.

petsild, 2016-07-19:
Found an older recording here: https://github.com/opentrack/opentrack/issues/321

combination  Pointracker/freetrack

sthalik, 2016-07-19:
@petsild I have a guess why it might have broken like that. Try https://www.dropbox.com/s/0e3r3g0fkr00qd1/opentrack-test-for-petsild-20160719.7z

If it's still broken, confirm it breaks with dynamic pose resolution **off**.

petsild, 2016-07-19:
I'm sorry I couldn't test before. 
Now is the octopus nervous and the size of the point itself jumps 2,65 – 2,90 px. Manual setting of the threshold does not affect.

petsild, 2016-07-19:
The smaller the size of the point I registered, original P8 4.2 and the current P9 to 2.9 px.

sthalik, 2016-07-19:
@petsild OK point extraction used wrong image as input. See https://www.dropbox.com/s/o9mpcnzla8jcivv/opentrack-test-for-petsild-20160719_2.7z

petsild, 2016-07-19:
I tested it half an hour and all the problems are probably eliminated. Thank you great job hi. Pet

sthalik, 2016-07-19:
@petsild glad to hear!

closed 2016-07-19 by sthalik