opentrack/opentrack #392: Feature Request: Camera Pitch in Source Settings

Spectator6, 2016-07-19:
Hello,

Excellent work with Opentrack, thank you for actively supporting this great software!  Would it be possible to allow the user to set something similar to the "Camera Pitch" angle in Facetracknoir?  [Image](http://imgur.com/Y23uxpW) camera is situated at the base of my monitor so the yaw interpretation is a bit "skewed" in Opentrack. In comparison to Facetracknoir, I've found that I can just about correct for this "angle distortion" by adjusting this setting to 30-35 degrees. Would be awesome to have a similar functionality for Opentrack. Here's an 12th century renaissance oil on canvas believed to be an artist's interpretation of a famous marble sculpture depicting this same idea [Image](https://imgur.com/OGzLLgG)

All the best, thanks again!

sthalik, 2016-07-19:
@Spectator6 go to options -> camera and specify the offset.

Spectator6, 2016-07-19:
Thank you for the quick reply sthalik!  

To follow up, I messed with the offset options per the manual, but I can't seem to achieve a similar result. I've attached two super short video clips to illustrate. It's quite possibly something I'm doing wrong on my end though!

Here's the initial clip with game data zeroed at start. Note how the direct lateral head turn is "seen" in an ellipse, as expected: 

[Zero at center](https://gfycat.com/ImpressionableTemptingFritillarybutterfly)

To compensate for this, I first looked to the side, then adjusted to bring the game data values for pitch and roll to neutral, more or less. When centering head position and turning the opposite direction, the ellipse appears still be present, just on a modified plane. 

[Zero pitch/roll at turn](https://gfycat.com/FewFatherlyDassie)

For comparison, FTNOIR: 

[FNOIR compensation](https://gfycat.com/WanRingedAmphiuma)

Let me know if you need any more info! All the best

sthalik, 2016-07-19:
@Spectator6 from your description it looks to me like only translation needs to be affected by camera offset, leaving rotation alone. Is it sensible to you?

WF000, 2016-07-20:
Hi, Spectactor6,  it seems to me that your model size and/or calibration is not valid.
-that is why the movement can be somehow an elipse , I've had similar case

Please try to re-measure your model and recalibrate afterwards, then we'll see.
Regards

sthalik, 2016-07-20:
@WF000 I've got similarly bad distortion caused by camera offset unless commenting out rotating rotation by the offset. Apparently only translation should be rotated by the offset.

Spectator6, 2016-07-21:
That did the trick, thank you sthalik!

For anyone else that may be wondering, I didn't realize that centering the camera with keybinds was different from setting the offset. Once I did exactly what was described in the manual without first centering by keybind, it worked like a charm :)

sthalik, 2016-07-21:
@Spectator6 does camera offset introduce bad rotation behavior for you?

closed 2016-07-21 by sthalik

sthalik, 2016-07-21:
@WF000 thanks. I'm leaving offset on for rotation also.

WF000, 2016-07-21:
maybe it could come in handy sometimes