opentrack/opentrack #393: Track Logging

DaMichel, 2016-07-19:
I want to discuss logging of the tracking data for development purposes. The goal is to write a time series of poses at various stages of processing to a file.

There should be an option to enable it, obviously. I actually changed the UI files already, and propose the following
[image]

My implementation proposal stipulates a logging base class which does actually nothing.
```
class TrackLoggerNop
{
public:
     // dummy methods that do nothing to not waste cycles
    virtual write(const Pose &pose);
    virtual next_line();
};
```
and a derived class that is used when logging is enabled.
```
class TrackLogger : public TrackLoggerNop
{
     std::ofstream out;
     QString filename;
public:
    // proper logging
    // writes csv formatted data
}
```
I think a proper place to instantiate it would be in ```void MainWindow::startTracker()```.
It should check for errors. In case the file could not be opened for writing a message box should appear. And possibly the tracking should start without logging.
The logger instance will then be handed over to the ```Tracker``` instance. So in ```void Tracker::logic()``` we will have statements like.
```
logger->write(pose); // raw, before filtering, after filtering, final output with mapping, etc ...
```
Ideally I would like to have everything contained in ```Tracker``` but it is not allowed to open a ```QMessageBox``` from a thread. Either way, either the filename setting or the file handle needs to be transported into the ```Tracker``` instance.

sthalik, 2016-07-19:
@DaMichel I recommend put the checkbox and the pathname into `struct main_settings`. Then, connect a signal to both of these to reinitialize the logger. Protect the logger with a mutex or use a `Qt::QueuedConnection` to `class Tracker`.

The rest is fine. Think up what stuff you need, e.g. dt, translation compensation result, etc.

sthalik, 2016-07-19:
Actually `class Tracker` has no Qt event loop. I recommend using a mutex so that changes are applied immediately.

DaMichel, 2016-07-19:
Okay, thanks for the hint.

DaMichel, 2016-07-22:
And there is the result. 
https://github.com/DaMichel/opentrack/tree/logging
In this implementation, settings take effect only when the tracker is started. To make this clear, I greyed out the relevant parts of the options dialog while the tracker is running.

No need to pull it if you don't need it. It needs a rebase anyway.

sthalik, 2016-07-22:
@DaMichel I added line comments.

As for signals, check out `options::base_value`. It has all the stuff.

closed 2016-08-04 by DaMichel