opentrack/opentrack #394: Roll affects pitch and vice versa (IK this has been mentioned before but your dropbox links are dead)

Cybotive, 2016-07-22:
I'm having problems with my roll affecting pitch and vice versa. It's very annoying and I'm using this in a VR setting, so I can't have that at all. I usually have my phone tethered over USB and point my phone to the right IP that my phone gave it, but I've also debugged with a wireless connection.

Speaking of debugging, when I turn on debugging and watch the orientation values while i pitch the phone only one of the three numbers separated by a comma moves but according to OpenTrack on the PC side, all three "Raw" values change significantly (mostly roll and pitch).

sthalik, 2016-07-30:
The freepie side uses Android framework functions to compute the actual pose. It doesn't perform centering.

freepie transmits euler angles. This is in accordance with the freepie cell phone API so that you can use a pristine freepie apk. The apk is maintained as part of the freepie project.

If you're having gimbal lock then use the "angle offset" function in our freepie tracker settings. It should help with the typical cases.

Cybotive, 2016-08-01:
Nope. Doesn't help at all. just rotates the octopus by the offset as expected, but the axes are still tied. I have confirmed it is opentrack by turning off "center at startup" and watching raw data. when it's flat on a table and i pitch up it only affects roll, and roll affects pitch, (so I swap them) but they're not tied in raw data, only calculated data no matter what filter i use.

sthalik, 2016-08-05:
There are two centering modes in the options window. Are you using the one for inertial trackers?

Cybotive, 2016-08-05:
Yes, I tried both. I also recently tried unchecking "center at startup" and it behaves the same way. Vridge and every app I have using the Cardboard API or Google VR SDK works, besides when using it with this.

sthalik, 2016-08-05:
Can you also check rc10's correct matrix -> euler formula?

sthalik, 2016-08-05:
If this fails, give me starting and ending raw values that give this bad behavior.

Cybotive, 2016-08-05:
Where/what is the correct matrix? I am not able to test immediately as it is 2:32am, but I will soon.

Cybotive, 2016-08-06:
It's still very wonky.

sthalik, 2016-08-06:
Give me the zero value of raw rotation and the rotated value that doesn't work for you.

Cybotive, 2016-08-06:
I figured I would just show it in action, the first half of the video, the values are just swapped so roll is affecting yaw (although settings are all good, no swapping) but that's easy to remedy. But the real problem is moving around in other directions.

https://www.youtube.com/watch?v=8xdr5QDANQQ
Wait a bit for 1080 to process if you see this right away.

thiagokrieck, 2016-08-08:
I am having the same issue while using pointracker. There seems to be some bleed of one axis into another.

sthalik, 2016-08-08:
This is harder than it seems. If someone can provide a geometrical solution, be my guest.

sthalik, 2016-09-21:
In rc12 we're using a different formula for euler -> matrix and matrix -> euler conversions. Also there's a third centering option that centers each axis ignoring all the others. Please confirm whether either of this solves the problem.

thiagokrieck, 2016-09-27:
For me the problem is completely gone.

sthalik, 2016-09-28:
This is consistent with other reports. Thanks for the confirmation.

Cybotive, 2016-09-28:
I'll have to try it out. I'll see if this is better in terms of drift than RiftCat alone.

sthalik, 2016-09-28:
@Cybotive report back if this is fixed please.

Cybotive, 2016-10-06:
Sorry for not testing. For me it hasn't changed, or maybe got worse.

Cybotive, 2016-10-06:
using 2.3-rc99p13

sthalik, 2016-10-06:
@Cybotive do raw values change for you when roll affects pitch, or is it only output values?

Cybotive, 2016-10-06:
it seems the raw values don't suffer from this, at least as much (shown on the left in OpenTrack).

Cybotive, 2016-10-06:
laying it landscape, flat on a table and leaning it forward and back, the only raw value that changes is roll (the others are +-2-ish). But with the calculated values, roll and pitch move quite a lot, and very minimally does yaw move, but still more than raw.

Cybotive, 2016-10-06:
but even in raw, when in portrait and leaning it around, the numbers skip at a certain point.

sthalik, 2016-10-06:
The IMU app that I've co-written uses Android internal functions for pose estimation. It's gyro+mag+acc integration.

You may want to try out other camera centering modes -- "experimental camerra" for instance.

Cybotive, 2016-10-06:
hmmmmmmm.

sthalik, 2017-01-16:
Magnetometers are in general total crap if anywhere near speaker or headphone magnets. Try moving them out of the room for testing to eliminate possibility of interference. Merely plugging them off won't work of course here.

sthalik, 2017-01-16:
If there are any more bugs related to that please send a message in this issue and I'll reopen it.

closed 2017-01-16 by sthalik