opentrack/opentrack #395: Feature Request: Question about multiple trackers (aruco) positioning

Tux0Topo, 2016-07-25:
I'm not sure if this is related to opentrack itself or to aruco develop...

Top view:

                 Aruco 
Aruco      HEAD ->    Aruco                        <- cam
                 Aruco

Is there any chance to set the center position of the head, away from the trackers? That way, the same aruco code in different planes of a figure, can track the center position of the figure (yaw, pitch and roll are tracked from different method). So, if i rotate the figure, without moving it, the tracking shows no positional change.

The same idea, but with added features:

                   Aruco #3
Aruco #2      HEAD ->    Aruco #1                         <- cam
                   Aruco #4

I hope the diagram can clear the idea. Multiple Arucos, indicating planes of the head (front, back, sides).
So, when the cam detect any code, can detect and identify what side of the head is looking, and make a better detection for rotation, also calculating the center position away from the markers.

I have tested the first idea, with only 3 trackers on the sides of a box, and that way i get better detection and tracking than using just 1 tracker. Sadly, cant find a way to set the center of the cube away from trackers, so when i rotate the box, the followed tracker show some movement in x & z coordinates.

sthalik, 2016-07-25:
It's not the matter of the center. It's lackage of the ability to input multiple markers and their 3D coordinates.

Your usage scenario makes sense but there's no code supporting that. If someone writes the code I'll review it.

Tux0Topo, 2016-07-25:
Thanks for answering!
So, different trackers is out of the table...

What about the same tracker away from center?

sthalik, 2016-07-25:
More trackers aren't out of the question.

The code needs to be written. Time is money.

closed 2016-09-21 by sthalik