opentrack/opentrack #396: Feature Request:Combine heitre plugin+1point IR tracker (for moves)

drdanilov21, 2016-07-26:
Hi all! 
It'll be intresting to combine arduino for detect Yaw Pitch and roll, and 1point IR tracker for X Y Z moves.
Is it real to combine 2 plugins in same time?

HMRMike, 2016-07-26:
But, how will we measure translation from one IR point?

Tux0Topo, 2016-07-27:
Unless you use something bigger, like a bulb (ps move controller? ;)  so the cam can calculate the diameter all the time for the depth, the just one led idea is useless...
What are you planning to use? Apps like RiftCatYaw Pitch and roll directly from cellphone, and X Y Z moves from opentrack...

drdanilov21, 2016-07-27:
ITs wrong to use one IR to Z moves, but I think, its possible to X Y moves, isnt?

Tux0Topo, 2016-07-27:
For just  2 coordinates, should be just fine. The actual coding should be incompatible to that, but can be done.
Actually, if you search for RevolVR controller program, thats what it does.... Track a dot (big one) in the plane of the cellphone camera, but not depth.

sthalik, 2016-07-28:
see #171

closed 2016-07-28 by sthalik