opentrack/opentrack #399: QFileDialog::getSaveFileName freezes sometimes

DaMichel, 2016-07-30:
Continuing the discussion ...
```
int main(int argc, char** argv)
{
#ifdef _WIN32
    const HRESULT hr = CoInitializeEx(NULL, COINIT_MULTITHREADED);
    if (FAILED(hr))
        qDebug() << "failed CoInitializeEx" << hr << GetLastError();
    add_program_library_path();
#elif !defined(__linux)
...
```
causes the freeze to occur perfectly reproducible by going straight for the file select button after launch.

sthalik, 2016-07-30:
@DaMichel what if you use `COINIT_APARTMENTTHREADED` instead?

DaMichel, 2016-07-30:
Yeah this is supposed to be the right flag. It is what I use now. Will keep it this way and report back if it freezes again.

sthalik, 2016-07-31:
Are you using coinit... in main with the native dialog? Then it makes sense, yes.

sthalik, 2016-07-31:
@DaMichel can you try removing TerminateProcess() on MSVC++ and seeing if it crashes on program exit?

DaMichel, 2016-07-31:
No. Crash. I'm going to use the following patch for a while and see what happens ...
https://github.com/DaMichel/opentrack/commit/777942eb65884ba9248c01399a4a5d1e72a35d80

sthalik, 2016-07-31:
It crashes after main ends. Can you get any meaningful backtrace out of it?

sthalik, 2016-09-26:
MSVC crashes in atexit. For now we terminate the process code with zero exit status.

We never use COINIT_MULTITHREADED in opentrack or opencv fork.

Another thing is calling `CoInitialize(nullptr)` rather than `CoInitializeEx(nullptr, COINIT_APARTMENTTHREADED)`. The latter messes up all of the threads' apartments. We should be all good.

closed 2016-09-26 by sthalik