opentrack/opentrack #402: Trackir 5 camera not detected by opentrack

MigLeader, 2016-08-01:
hello opentrack developers , i have an issue as the title says , opentrack cant detect trackir 5 camera , even though i have disabled my internal camera , my specs are as follows:

win 8.1 64bit
MSI GS60 2QE (I7 4700 , GTX970m , 16GBram)
Trackir 5

i don't know what other info i must provide , but i will be gladly provide anything

thanks for any help i get.

[image]
.

Len62, 2016-08-02:
Could be that the camera is not identified as a camera in windows. Since it was designed to work with their software it may be hidden to other apps. It should be listed under the "imaging devices" in order for windows to see it?

sthalik, 2016-08-02:
opentrack only supports directshow cameras, not proprietary camera protocols. Take a look at linuxtrack.

closed 2016-08-02 by sthalik