opentrack/opentrack #403: Report(white fang)2.3rc49p9-38-g26ab8a2

WF000, 2016-08-03:
Hello,
Testing this version shows good/tolerant rotRz improvements 
very, very small pitch, raw:+/-5 , 
used MANUAL calibration, default accela settings
-strangely it seems there is some kind problem with offset correction related with pitch/rot connection ,  when I've input correction for offset - problem Rz/Ry problem occurs
-if no offset correction , no Rz.Ry problem
If you need more , I'll record a video

sthalik, 2016-08-04:
Translation compensation and the experimental option of translating by the center value are mutually exclusive. This hasn't been fixed yet.

Manual camera offset is counterintuitive, I agree.

Try not centering at all and see how it changes axis interconnect. I'd like to know whether centering is necessary for breakage.

Do you say that there's more incorrect roll than in last release version?

WF000, 2016-08-05:
Hi, no.  much smaller inter-connection with roll then before, better then before.

Ok, with no center on startup resulted like this (used default settings on everything then axis mapping, dead zones etc.). all values in raw:
Rz axis is increasing+ in any and both cases:  Ry up and/or Rx left
-moving up and left rotates to the left and moving down and right  wants to rotate to the right
-so any of this movements, and both, in those directions causes Rz to rise+, when movement in opposite direction Rz lowers- but not greater then +/-<10 , so smaller values then before

So it is like for 4 points of Ry up you get 1 Rz , and for -7 points of Rx left you get 1 Rz also.

Also, no, I dont think centering has any effect on this, same difference in values and after centering.

sthalik, 2016-08-05:
People usually complain about pitch being connected with yaw and roll. It's also from my experience that yaw/pitch interconnect is the worst.

sthalik, 2016-08-05:
When you disable centering, is it any different from the raw values?

WF000, 2016-08-05:
No, same behavior, difference, observed with center on/off , and after manual centering.
So, no centering has not any effect on this, means, centering is ok, in my observation.

sthalik, 2016-08-05:
That's not it. Are mapped values visibly different from raw values?

WF000, 2016-08-05:
Well I'm not sure what do you mean exactly, please try to put it another way.

I've just set axis mapping to be non-linear and to include dead-zones, especially bigger for roll,
but the values of output mapping are ok, proportional to raw input. Was that your question?

sthalik, 2016-08-14:
Yes, that was an answer to my exact question.

It should be possible to de-interconnect all the axis but the geometry eludes my simple understanding. There's some work already done on the issue however.

sthalik, 2016-08-27:
I've got geometry code that works.

sthalik, 2016-09-04:
Test build(s): https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa

closed 2016-10-05 by sthalik