opentrack/opentrack #404: Question for aruco input with ps3eye cam @75hz...

Tux0Topo, 2016-08-03:
So... i was testing the ps3eye cam. By far, better for tracking that any other common webcam....
So, i start to try with freetrack app, and could set the speed to 75hz.
Then choose the opentrack, with aruco, and only could set it to 60hz... Then, changed the input to freetrack and voilá, 75hz again...
I'm doing something wrong? or the aruco input plug in, is limited to 60hz?

Len62, 2016-08-04:
Open the aruco settings and you'll see the options available.
[image]

sthalik, 2016-08-04:
Confirm that copying cleye.config to the "modules" subdirectory fixes the problem.

Tux0Topo, 2016-08-04:
Len62: Yes... but the Frames per second, only show Default, 30, 60, 120, 200... but no 75...

sthalik: The cleye.config is in the main directory...can't find modules subdir...
Inside the file, only this:
<?xml version="1.0" encoding="UTF-8"?>
<cleye>
    <item name="mode" value="advanced" />
</cleye>

Tux0Topo, 2016-08-04:
Not sure if related, but i'm on RC7...

sthalik, 2016-08-04:
You can set more FPS than allowed and it'll use 75. But 60 has less lossy compression artifacts.

I guess cleye.config may have Unix line endings and if you're opening it in notepad++ then it won't show all lines. It should look like

    <?xml version="1.0" encoding="UTF-8"?>
    <cleye>
        <item name="mode" value="advanced" />
    </cleye>

sthalik, 2016-08-14:
All seems to work fine. Set fps to 200 or more. The PS3 Eye driver will adjust to max fps for that resolution.

Also I've added 75 Hz to the aruco tracker dialog.

closed 2016-08-14 by sthalik