opentrack/opentrack #405: P10  automatic threshold

petsild, 2016-08-06:
Hi,
the last P10 when a friend Leon turns on automatic threshold green crosses are turned off and are only white points. 
The range of the size of the points has no influence at all.
The manual works well. 

HW:
Model Cap 
Camera PS3 640x480@75
System Win 10

Len62, 2016-08-06:
Interesting. Automatic threshold is working fine here w/P10. I have the same specs except I use a clip. What kind of pixel sizes does your friend get using manual?

sthalik, 2016-08-06:
@petsild what's your point area size?

petsild, 2016-08-07:
shows 15px

sthalik, 2016-08-07:
@petsild max value is 10 px. Can you try if lowering exposure makes it work, just to make sure? If it does then I'll put the max value at 16 px.

Len62, 2016-08-07:
Even with auto threshold selected the threshold slider may need to be adjusted.

sthalik, 2016-08-07:
@Len62 what I meant is that the slider's value most to the right needs to get changed from 10 to 16. I'd like to know how it works with lowered exposure (hence smaller points), first.

sthalik, 2016-08-07:
By the way I have a really dumb way to fix gimbal lock once and for all for all devices.

sthalik, 2016-08-27:
Max radius is 15 now.

petsild, 2016-08-27:
If I have a point in the neutral position of 15px, and then I zoom in on the camera? For sure will 20 value better?

sthalik, 2016-08-27:
No, it makes small values hard to adjust.

sthalik, 2016-09-04:
Test build(s): https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa

petsild, 2016-09-04:
When you try to run it show error.

[image]

Len62, 2016-09-04:
Strange, running fine so far here. Have yet to really try it out.

sthalik, 2016-09-05:
@petsild do you have the "platforms" directory and qwindows.dll when unzipped?

petsild, 2016-09-05:
Friends it also does not start. Always OT worked now error.
The whole time running f:\opentrack
I need something else to install?
Thank you for the support.  Pet

sthalik, 2016-09-05:
I see the problem.

Get today's second build. http://goo.gl/Fq8CT7

sthalik, 2016-09-05:
Can someone try start opentrack on Windows XP, please?

petsild, 2016-09-06:
I tested the latest build for download:
[image]
I can run but all the curves in the profiles are lost! The new version will require setting the new curves?

Len62, 2016-09-06:
Yes, probably due to changes in how mapping works.

sthalik, 2016-09-06:
Import function's not yet written.

sthalik, 2016-09-06:
For the record, rc12 will import splines as long as no new one have been set.

petsild, 2016-09-07:
Automatic threshold still doesn't work.

sthalik, 2016-09-08:
Decrease exposure so points aren't larger than 15 px.

petsild, 2016-09-10:
Tested on the size of 12px the actual size of the point. The change of scope has no effect.

[image]

Len62, 2016-09-10:
Why is the video window so bright and noisy? You may want to play around with the CLeye driver to get your picture darker and less noisy. Are you using IR leds, and if so have you removed the IR filter from the camera?

petsild, 2016-09-10:
AT the verse P9 all work on the same light configuration as in the picture.

Len62, 2016-09-10:
Ok. Just a thought. You do know that with automatic checked you can still adjust the slider right? The slider works with auto on or off, it just functions differently.

sthalik, 2016-09-21:
@petsild it works when I set exposure and gain as high as this. What happens when you set min size to 0 and max size as 999?

sthalik, 2016-09-21:
@petsild it makes more sense to use very low exposure and gain values so the light's center area is brighter than the edges. Then PT gives more weight to the brighter parts but doesn't discard the rest.

What I did was lower color balance a lot, set lowest exposure and slightly adjust gain. Then barely anything is visible and the LED is visible as different shades of gray, not a single white flash. But I also have several unexposed film layers on the sensor so even IR could be darker than normal.

sthalik, 2016-09-21:
My single LED, zoomed in. Auto threshold shows 6 px.

[image]

sthalik, 2016-09-24:
@petsild run the experimental build. I forgot to multiply by `pi` for point area.

https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa

petsild, 2016-09-24:
Features AT again works flawlessly :) to all my friends. 
Thank you very much for the patch, great work hey. Pet

closed 2016-09-24 by sthalik