opentrack/opentrack #406: Microsoft FSX SimConnect problems

MathijsG, 2016-08-06:
I finally have a new graphics card today, and I wanted to play some Prepar3D.

However when I want to load the FSX Simconnect I get this:
[image]

This is with the newest version. No matter what version I select in settings.

sthalik, 2016-08-06:
Install SimConnect SDK and its updates that come with P3D. If it still occurs, send me the P3D simconnect manifest.

MathijsG, 2016-08-06:
It does already work with SimConnect protocol in FaceTrackNoIR, but I will check further with the SDK.

MathijsG, 2016-08-07:
I thought the Microsoft FSX SimConnect protocol should work even without having any of these simulator games installed (I do have though). Or am I mistaken? It seems rather weird that in FaceTrackNoIR the protocol works perfectly.

sthalik, 2016-08-07:
@MathijsG send me the simconnect dll from p3d.

sthalik, 2016-08-07:
It always requires the sdk, and there are multiple sdk versions that require "SxS manifests", an annoying windows feature. ftnoir does it similarly to how we do it, except opentrack allows for multiple SDK versions (e.g. plain FSX, SP1, SP2...).

MathijsG, 2016-08-07:
Prepar3D V3 has different SimConnect's it provides. I will give you all of them.

You can download them here (they are inside their respective MSI file).

https://mega.nz/#!rUZQhDwC!RldVQITeu3VbRkAobaEvSHG72um0BuoBgf_eSsHZxQQ

sthalik, 2016-08-07:
Okay then you should be able to install SP1 and select SP1 in protocol settings.

MathijsG, 2016-08-07:
I already tried that, tried all of them. But I still get that message all the time.

MathijsG, 2016-08-08:
Even trackhat got it working with no problems. Can there be something wrong with this in this build?

MathijsG, 2016-08-08:
Tried the following:
- Uninstall P10. reinstall P10.: Didn't work.
- Remove every SimConnect version and try one version and test that: Didn't work
- Try TrackHat: Did Work (1.2p2)
- Copied the simconnect tracker DLL from TrackHat to Opentrack P10: Didn't work
- Removed Opentrack P10. Installed P7 (latest version with the setup still downloadable). Worked immediately.

I guess there is something wrong with the latest build in terms of simconnect?

sthalik, 2016-08-08:
Yes, I got it. Dumb Windows SxS requires accurate .dll path.

sthalik, 2016-08-08:
See if this works - https://www.dropbox.com/s/gm4mbfrwg8ruwx2/opentrack-simconnect-test-20160808.7z

sthalik, 2016-08-08:
The reason for breakage is moving the plugins to the "modules" subdirectory.

MathijsG, 2016-08-08:
Ah, that idea had crossed my mind already.

This one is working without problems.

So if I understand correctly: Even if you don't have SimConnect installed on your computer this library load error should not occur? It only would just not work then, because of the absence of SimConnect istelf?

sthalik, 2016-08-08:
This is the same error but for the wrong reason :)

Windows is so dumb that the expected manifest needs to be a resource, just like an icon, inside the specified file. So we have several numbered manifests in the .dll.

closed 2016-08-08 by sthalik

jpslara, 2016-08-12:
I have the same problem with FSX from steam, with the error saying that a lib can't be loaded.

I hab seen  that installing opentrack version 7 will work, and i tray it, and it work for me too, with opentrack 7 work, with opentrack 10 don't work.

sthalik, 2016-08-12:
@jpslara 1) install simconnect sdk 2) use rc7 for now 3) select SP1 in protocol settings

commit 39e63374 (2016-08-08): proto/simconnect: specify right path after hier(7) changes

Tested-by: @MathijsG
Closes #406