opentrack/opentrack #407: Vjoy Demo Expired error

Pure-Awesomeness, 2016-08-06:
Latest version of opentrack causes vjoy output to display a message box "Demo expired" after a couple minutes of use. Drivers were downloaded from the headsoft website linked when clicking on the options button. Is this a bug? How can I fix it?

sthalik, 2016-08-06:
I'm going to replace vjoy with an open-source joystick driver. This is a problem definitely.

Pure-Awesomeness, 2016-08-07:
So is there a temporary fix for this?

PS: Sorry for your cat's loss, Keep up the great work sir!

sthalik, 2016-08-07:
Temporary, no. The vjoy protocol needs to be replaced with another driver.

sthalik, 2016-08-08:
See if https://www.dropbox.com/s/3ygvghboe7kk2p2/opentrack-vjoystick-20160808.7z works. You'll get instructions on how to install the driver.

Pure-Awesomeness, 2016-08-08:
Works very well. Thanks

sthalik, 2016-08-08:
Thanks for the confirmation.

closed 2016-08-08 by sthalik