opentrack/opentrack #408: When viewed at 90 degrees, the offset is displayed incorrectly

olegiy, 2016-08-08:
Hi all. 
Previous version opentrack-2.3-rc49p7 --> https://youtu.be/gY9XFM_E5RY
Latest version --> https://youtu.be/P54s4nAS2Ig
The settings are not changed.

sthalik, 2016-08-08:
What offset?

olegiy, 2016-08-08:
displacement of the head to the left and right

sthalik, 2016-08-08:
If you move 90 degrees to the right, moving left should move you toward the front of the aircraft. Is it working correctly?

olegiy, 2016-08-08:
https://youtu.be/uFjrXqIGlnc

sthalik, 2016-08-08:
I cannot reproduce. See if https://www.dropbox.com/s/z1f6u5voc0lz1af/opentrack-tcomp-20160808.7z works for you, it's the latest dev version.

sthalik, 2016-08-08:
And don't enable "use center as translation offset" since it works badly with compensation.

olegiy, 2016-08-08:
Thanks for patience. I will test program tomorrow. And than I will write you.

olegiy, 2016-08-09:
The latest version better, but not as rc49p7. The range of movement is less.. 
"use center as translation offset"  - disabled. 
(translate.google) © )))

sthalik, 2016-08-09:
Are you referring to translation max value being 50? What max value do you need for War Thunder?

sthalik, 2016-08-09:
If that's the case, do you need the input and output to be more?

olegiy, 2016-08-09:
65-70

sthalik, 2016-08-09:
But output or input?

olegiy, 2016-08-09:
output

sthalik, 2016-08-09:
@olegiy confirm it works. https://www.dropbox.com/s/mj1lsd3604c3npa/opentrack-tmax-20160809.7z

If you still have doubts regarding translation compensation, please take your time until you're sure.

sthalik, 2016-08-14:
@olegiy is it working now?

olegiy, 2016-08-14:
I will check only tomorrow.

olegiy, 2016-08-15:
All is well. Thank you.

closed 2016-08-15 by sthalik

sthalik, 2016-08-15:
@olegiy thank you for the confirmation.