opentrack/opentrack #409: add 75hz option to Aruco with PS3 Eye camera

andregm3, 2016-08-08:
Since PS Eye can work at 75hz there should be an option to use this frequency on Open Track Aruco:
[image]

Tux0Topo, 2016-08-08:
I'm sorry... JAJAJAJAJAJAJAJAJAJAJA
I've posted the same thing about a week ago.
Setting the frames to something like 125, will turn you to scan to the maximun speed of the camera (75hz). The bad news, seems to be that there will be some noise while tracking... That's why the best bet is to stay in 60hz...

andregm3, 2016-08-08:
yeah i realized that too for 60hz, but since i never been able to try 75hz option i couldn't know if was it because of the 125hz permission.

closed 2016-08-14 by sthalik

sthalik, 2016-09-04:
Test build(s): https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa

commit a95313c4 (2016-08-14): tracker/aruco: show 75 FPS option in combobox

Reported-by: @andregm3
Closes #409