opentrack/opentrack #410: rc49-p10 - minimize to tray doesn't work

albertolg, 2016-08-09:
previous version worked

Len62, 2016-08-09:
Isn't that normal behavior?

Len62, 2016-08-09:
It shows up on the task bar and in the tray when minimized here.

sthalik, 2016-08-09:
It's not. I'm fixing it.

sthalik, 2016-08-10:
@albertolg see https://www.dropbox.com/s/zcxqqqczx3qpghu/opentrack-fixes-20160810_1.7z

cheers

albertolg, 2016-08-10:
It crashes on launch

`Firma problema:
  Nome evento problema:	APPCRASH
  Nome applicazione:	opentrack.exe
  Versione applicazione:	0.0.0.0
  Timestamp applicazione:	57aae802
  Nome modulo con errori:	Qt5Core.dll
  Versione modulo con errori:	5.7.0.0
  Timestamp modulo con errori:	57788b91
  Codice eccezione:	40000015
  Offset eccezione:	0000e9ba
  Versione SO:	6.3.9600.2.0.0.256.4
  ID impostazioni locali:	1040
  Informazioni aggiuntive 1:	8f76
  Ulteriori informazioni 2:	8f76c0d18689ee02711cca42f268a735
  Ulteriori informazioni 3:	91d9
  Ulteriori informazioni 4:	91d969fda2dd6556c53187e0637314ab

`

sthalik, 2016-08-10:
Please copy the "platforms" file temporarily from rc10. I'm updating the file.

sthalik, 2016-08-10:
Updated.

Len62, 2016-08-10:
Okay, it minimizes to the tray correctly. However, you need to double click the tray icon to open it up, as opposed to just a single left/right click.

sthalik, 2016-08-10:
That's true. You want single left click also? Okay...

Len62, 2016-08-10:
Just did not know if that was your intent. I'm fine with it the way it is.

sthalik, 2016-08-10:
You have a point. Let's make both single and double click work.

How do you feel about the lack of tray menu?

Len62, 2016-08-10:
Maybe a tray menu that offers to open/exit program.

albertolg, 2016-08-10:
@sthalik there is a little problem...

1) minimize to icon
2) double click to show the window, works
3) minimize again
4) double click again, opentrack icon just appear in the taskbar but no window.. click it and it disappear again

sthalik, 2016-08-14:
@albertolg following the latest changes, I've tested the tray stuff extensively. It should work. It's the double click that makes it not show. Now that single clicks are used it toggle each time. I'll leave this bug open until you confirm that rc11 works.

albertolg, 2016-08-14:
@sthalik tested now, works but still a little insignificant issue... when starts it is minimized, perfect.. click the tray icon = maximized... ok... click again the tray icon, minimize, ok... but the issue is when you minimize the window from the minimize button.. the next time that you click the tray icon it doesn't maximize but appear only in the taskbar... then you need to click again the taskbar icon or the tray icon.. you can leave it as is and fix later, we can live with it :-)

sthalik, 2016-08-14:
It's done for the next rc.

sthalik, 2016-09-04:
Test build(s): https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa

Len62, 2016-09-05:
Minimize to tray working correctly and I really like the options available for direct access! So far everything looking good for me with this build.

sthalik, 2016-09-05:
Thank you for reporting back and the feedback.

closed 2016-09-05 by sthalik