opentrack/opentrack #411: Question about space for tracking...

Tux0Topo, 2016-08-09:
Hi!!
I was testing with aruco tracker, how far away i could go from camera, and found that the limits are 50/50/50 and -50/-50/-50 for coordinates. The cam output shows me that the tracker is still being detected out of those limits, but the coordinates are stuck in 50.... Those limits seems to be for about 1m^3 in front of the camera.
So, is there any way to make this tracking space bigger?

sthalik, 2016-08-09:
How much do you need for input?

sthalik, 2016-08-09:
I'd go for 75 since that's what input is. Are you fine with that?

Tux0Topo, 2016-08-09:
75 should be able to scan 1.5mt, right? I was hoping to get 2mt, or maybe better if can be selected by the user... That way, if get some better camera, optics, o better tracker can be changed directly by user...
Appreciated all the efforts!!!

sthalik, 2016-08-09:
@Tux0Topo I can also change aruco marker scale.

Tux0Topo, 2016-08-09:
Whatever you think that can give you easier work, will be wonderful.
Changing the aruco marker scale can be translated into bigger tracking space?

Tux0Topo, 2016-08-09:
@sthalik I've been testing the beta that you made for the 90 degrees thread, and it's fantastic. I think that the input size can be made a little bigger. And maybe the output turned to 100?

Thanks sthalik!! You rulz! :D

sthalik, 2016-08-09:
@Tux0Topo for most games even output of 50 would be excessive. It's only a few games that need 75.

Model size was 40, now it's 100. I'll make it 150.

sthalik, 2016-08-10:
@Tux0Topo see https://www.dropbox.com/s/zcxqqqczx3qpghu/opentrack-fixes-20160810_1.7z

cheers

Tux0Topo, 2016-08-11:
Hi sthalik!!
Sorry for the delay in testing, but been busy.
This last version, became crazy. It's a no go.
The sensitivity is un-usable. I move the tracker just a little, and pass from 75 to -75 in no time. Rotation have the same defect. The result is less phisical space to move, and too much sensitivity.
The previous "opentrack-tmax-20160809" was much better.

sthalik, 2016-08-12:
@Tux0Topo was I right to increase marker size rather than decrease it? I'm not sure which one makes translation values smaller. Maybe I should decrease it after all.

sthalik, 2016-08-14:
@Tux0Topo I'm deferring this issue since I'm unable to figure it out on my own without a marker.

Tux0Topo, 2016-08-14:
No problem @sthalik! Thanks for the support!

sthalik, 2016-08-14:
@Tux0Topo so the later builds got too little space for translation? The original marker size was 40. Then 100, then 150. Bigger the number less space for translation, right?

sthalik, 2016-08-15:
@Tux0Topo I doubled the range for X and Y. Just keep in mind, the tracker is meant to be used sitting without moving around the room, as all others.

Build - https://www.dropbox.com/s/dkc58d5op4wwqi0/opentrack-aruco-translation-20160815.7z

closed 2016-09-26 by sthalik