opentrack/opentrack #412: [Feature] Ability to set the range of the translation graph

MathijsG, 2016-08-12:
Every game handles the arbitrary values of the X Y Z axes differently. I recently started to play X-Plane which is a popular multiplatform flight simulator. However I encounter that the minimum values you can pick in the grap which is 5, is already way too high for the max range in X-Plane. I think this game may be quite an exception, but it may come in handy for people to be able to change the minimum value and max value, instead of a fixed 0 - 50 scale. Maybe you even want to be able to make it a higher resolution, instead of being able to only select 5, 10, 15, etc. Maybe you want to have a range up to 20 for example, where you can individually select 1,2,3 etc.

Is this somehow possible in an easy way? I will check whether there are more games that don't work well with the current range.

sthalik, 2016-08-12:
Technically, yes. Adding new options needs careful consideration however.

Also, use PilotView. The default plugin introduces artificial rotation delay.

sthalik, 2016-08-12:
Would you rather have the graphs change their numbers or a multiplier? I like the former.

MathijsG, 2016-08-12:
I will check this PilotView thing you say. Maybe that already is a solution. Because X-Plane is for now the only game that handles the axis values this extreme. X-Plane itself looks like a horrible product to me (sorry developers :-P) so maybe it's just X-Plane.

A Multiplier will change the sensitivity and changing their numbers will just increase the range? Sorry not always that technical :(.

sthalik, 2016-08-12:
These sims like FSX are dependent on payware for any real usage. But from my interactions, the devs are competent.

A multiplier could change the range. But the range could also be explicitly changed at runtime.

MathijsG, 2016-08-12:
With this PilotView the range is already within your graph, although it feels a little more jerky than the native TrackIR module, but I also don't have many FPS in X-Plane 10, so maybe it's because of that.

What if the top and bottom and/or left and right most inputs are text-inputs, then it won't make many clutter.

Like this
[image]

And maybe also on the other axis range, but you get the idea?

sheepsskullcity, 2016-08-20:
I vote for this feature. But please don't set axis range limits to low. For example kinect have trackable range of 6m. So it needs 300 units of freetrack range.

sthalik, 2016-08-20:
@MathijsG you can disable PilotView artificial shaking. But it has real movement (including roll) as specified by opentrack's filter. Native is delayed on purpose.

I won't put a selector inside the mapping widget. It'll break with different OS (macosx has differently-styled widgets) and HiDPI. So aligning it pixel-perfect is pretty much impossible. Unless it's on the very bottom on its own or something.

I don't have a problem with the feature - in recent build the mapping widget and the map itself already use the options api exclusively, at the cost of a negligible slowdown.

sthalik, 2016-08-20:
And to be clear - it's possible of course to get instant update of the widget, and make "cancel" annull all the changes to the scale.

MathijsG, 2016-08-21:
What would you propose then for changing the values/changing the scale? I understand the thing about different OSes, but what would be a solution?

sthalik, 2016-08-21:
There's tons of empty space at the bottom of the widget. Let's minimize unneeded empty space, and add space for widgets.

sthalik, 2016-08-21:
That is, the bottom axis map needs stay visible and not cropped.

sthalik, 2016-08-21:
Now that I think of it, the "asymmetric" button can go somewhere also.

DaMichel, 2016-08-23:
I would highly appreciate configurable axis ranges on X and Y. Thanks for your consideration @sthalik .
The reason is, I mostly need input ranges only up to 20 degree or so. But the graph goes to 180 degree. Therefore adding a small non-linearity becomes very fiddly.

sthalik, 2016-09-05:
@DaMichel @MathijsG is it enough that we have tooltips now?

sthalik, 2016-09-05:
@MathijsG (try today's second experimental build).

MathijsG, 2016-09-16:
Will try further. But won't we move the problem by just increasing to 70 instead of 50? Because these values are arbitrary for the translation axes (X, Y Z) there could be easily another game requiring 80 for example.

Keep up the good work, I will test the newest RC for you now.

sthalik, 2016-09-16:
@MathijsG typically freetrack protocol games only use up to 20 of the entire range. Making it 70 is an okay compromise. If there are further complaints about the range being too small then I'll think up something better.

sthalik, 2016-09-16:
@MathijsG we can now resize the graphs like magic, even if the size is specified in the options window. Even having both options and mapping windows open and switching profiles will load everything so that no user interface elements have stale data. This is new in rc12. Previously some things stood still despite profile changes. You can see how tray checkbox isn't applied immediately.

sthalik, 2016-09-16:
@MathijsG are your mappings migrated properly? As well as "asymmetric axis" checkboxes and modules used.

closed 2017-02-27 by sthalik