opentrack/opentrack #413: Axis-mapping broken for me on 2.3-rc49p10

Creat, 2016-08-13:
Hello,

I've just reinstalled OpenTrack for the first time in about a year, where I've spent a lot of time creating a PointTracker profile back in version 2.3-rc16. That by itself seems to be still working perfectly: the red + showing my head-center position is dead on when I move my head around. The problem is that the game-output doesn't seem to be properly related to the raw data and I can't seem to fix it either.

I have most axis disabled for the profile in the options, the only ones enabled are Yaw and Pitch (mapping to yaw and pitch, of course) and both happen to be inverted. When I turn my head to the side, I instead get Roll-axis game data (roll axis output is set to disabled). If I head to options -> output and disable the inverted-checkboxes for both yaw and pitch, only pitch reacts. The roll-values are negated by the checkbox for the roll-axis (which is disabled) though. See [this imgur album ](http://imgur.com/a/Ul562) for two screenshots of the options and a "pose", note the raw/game discrepancies.

I've managed to get something resembling the behavior I wanted by mapping roll output to yaw input and disabling yaw output. Confusingly the roll output is 0 with this and it clearly just confuses yaw/roll. Is it just that dialog and/or the associated axis mapping that's messed up?

As a side-note, my asymmetric pitch profile seems to have swapped upper and lower half, is that expected or has something changed in the handling of that? For example, is it no longer inverted but applied after that now?

This may or may not be related to #394, I really can't tell. He seems to be getting "crosstalk" onto the roll-axis, or it might just be sensor-drift, but in principle his motions translated correctly otherwise I think?

Thanks for any help!

sthalik, 2016-08-13:
Gimbal lock is already worked around in unstable. Wait for next rc. It actually works this time and it should be impossible to get roll with pitch and yaw even with both values at +-180.

Sign swappage got fixed in unstable.

Interconnect isn't fixed until someone submits necessary geometry knowledge.

Creat, 2016-08-14:
Thanks for the reply, I'll just revert to rc16 then until the next version with the fixes is out and check again.
Alternatively, is there any reasonably stable (but more recent) version that you would recommend instead?

In any case, thanks for all the work put into this, it really is appreciated :)

sthalik, 2016-08-14:
> Alternatively, is there any reasonably stable (but more recent) version that you would recommend instead?

Try rc7, not rc10.

sthalik, 2016-08-14:
Or try rc11 which has all the fixes I mentioned.

Creat, 2016-08-18:
Hi again, well since the version that included the fixes (rc49-p11) was out the next day anyway, I just used that. I assumed it would take longer, so thanks for that :)

Since my issue is resolved, I'm just gonna close this (but I'll open another ticket for an unrelated issue).

closed 2016-08-18 by Creat