opentrack/opentrack #415: rc49-p11 curves are broken

albertolg, 2016-08-15:
the asymmetric mapping are ignored, worked with p7

sthalik, 2016-08-15:
It works for me except you need to select the checkboxes again.

The reason for that is that options window and mapping window saved each other. Now they are separate in the .ini file.

sthalik, 2016-08-15:
@albertolg can you confirm what's happening, please?

albertolg, 2016-08-15:
@sthalik the problem is in the output remap... I use edtracker and need to remap the axis like in the picture, but with the p11 the remap is not saved, you remap, works.. change profile and open the previous where you changed the output remap and is not there

[image]

sthalik, 2016-09-04:
Test build(s): https://www.dropbox.com/sh/823dd4wyhixv5a7/AACB1Iqquvia_keD9jUY2qFBa

sthalik, 2016-09-24:
rc12 has migration support for options changing representation. Also remap is now saved like it should.

closed 2016-09-24 by sthalik